luni64/TeensyStep #86: STM32 CUBE IDE

lexoring, 2020-11-17:
Hello!
Great job! Could you help me? I am using STM32f411CE (blackpill) and development environment from ST CUBE IDE and CUBE MX. Tell me if it is possible to transfer your project to this development environment, or maybe someone had such experience.

RockerM4NHUN, 2020-11-17:
Hi! I have a fork where I made quite radical changes to fit the library for STM32Cube environment.
My library is not really compatible with the original anymore but maybe you find some useful code snippets.
Be aware that it is not finished, bugs may occur.
Also @ramboerik started an stm32 branch, with arduino environment, it is probably in better condition than mine.
I think you should consider switching to Arduino env if possible, so you can use erik's work.
Cheers.

lexoring, 2020-11-20:
Hi! Thank you so much! I started to study your work.

lexoring, 2020-11-22:
> Hi! I have a fork where I made quite radical changes to fit the library for STM32Cube environment.
> My library is not really compatible with the original anymore but maybe you find some useful code [...]

Hey. Thanks for your great work. I am now setting it up in cube ide. Introduced this code:

I am tuning one motor:

#include "STM32Step.hpp"

GpioPin motor1_step {
GPIOA, GPIO_PIN_1};
GpioPin motor1_dir {
GPIOA, GPIO_PIN_2};

GPIOStepper motor_a(motor1_step, motor1_dir);
motor_a.setMaxSpeed(1000);
motor_a.setTargetRel(1000);

But I don’t know how to start it, could you help me with this piece of code. I would be very grateful to you.

GPIOStepControl controller();
controller.move(motor_a);

I am currently trying to tune one motor. but I will need to control two stepper motors synchronously but at different speeds. Generally I make a winding machine for transformer coils.

luni64, 2020-11-22:
Maybe this worked out example of a high precision winder might help you:
https://luni64.github.io/TeensyStep/applications2/010_winder/winder The complete code can be found in the examples section. 

Generally, the STM branch is maintained by @ramboerik, I can't give you much advice on this. But, from the application point of view it should work like the Teensy version.

RockerM4NHUN, 2020-11-22:
@lexoring If I see correctly (from the STM32Step.hpp include) you try to use my implementation.
Sorry for being inactive on that, some uploaded examples would probably help, but I am swamped with work at the moment.

For winder implementation I would suggest checking out luni's link.
For porting it to STM32Cube project here is some help:
- you will find app.h+your_app_name.cpp example in STM32TimerArray examples/project_setup_with_cubemx
- you will also need to setup a TimerArrayControl from that library, so it helps to familiarize with it

Usage of multiple motors did not change, you can see luni's examples for that, they should work.

And some code that I used for testing STM32Step:
```
#include "app.h"
#include "tim.h"
#include "STM32Step.hpp"

// you have to setup TIM2 correctly in CubeMX, check STM32TimerArray lib how to do that
TimerArrayControl timerControl(&htim2, F_CPU, F_CPU/1'000'000, 32);
GPIOStepper xmotor(GPIO_PIN(X_STP), GPIO_PIN(X_DIR)); // GPIO_PIN is just a macro so I don't have to write it out all the time
PositionControl posCtrl(timerControl, 10, 5'000);

// call this from main.c through app.h include, just before the infinite loop.
void app_start(){
    uint8_t usteps = 16;
    int32_t speedTarget = 500*usteps;
    xmotor.setMaxSpeed(speedTarget);
    xmotor.setAcceleration(speedTarget/2);

    posCtrl.begin();

    xmotor.setTargetRel(125*usteps);
    posCtrl.move(xmotor);
    
    xmotor.setTargetRel(125*usteps);
    posCtrl.move(xmotor);
    
    xmotor.setTargetRel(-250*usteps);
    posCtrl.move(xmotor);
}
```

ramboerik, 2021-01-23:
I guess this issue isn't relevant anymore and can be closed. I have no experience with the cube IDE and cannot give any useful tips 😄

closed 2021-01-23 by luni64