luni64/TeensyStep #89: More than 4 groups of independent movement

jackey984, 2020-12-20:
Hi luni64,

Thanks for providing this amazing library! I am implementing TeensyStep to control my 6-axis serial-link robot. The objective is to control 6 motors so as to regulate the end point to follow a given path, such as a straight line.

For path following, your RoseFunctionFollower.ino seems a good example. However, current TeensyStep has limitation of 4 groups of independent movement. Is it possible to relax the constraint to 6 groups? I created 6 RotateControl for the 6 motors, but only the first 4 can move.

Thanks,
Jackey

luni64, 2020-12-20:
Unfortunately not, each group needs a PIT timer and the Teensies only have 4 of them.

jackey984, 2020-12-20:
Thank you for the info! I have tested one single StepControl to move 6 motors in sync with acceleration. It works great for point-to-point motion. But in this way, currently it cannot work for path tracking, which requires to change Target on the fly when the motor is running. 

If disable the acceleration regulation, is it possible to make StepControl to accept changing Target?

closed 2021-09-28 by luni64