luni64/TeensyStep #91: getCurrentSpeed() not working STM32

lmonteiro440, 2021-01-10:
Hi. I´m using your library with a nucleo-64 F401re and I got a error from compiler when using the function to get the speed that got the timer clk frequency from hardwire timer library.

Here is the msg that I got from arduino IDE:
Arduino: 1.8.13 (Windows 10), TD: 1.53, Board: "Nucleo-64, Nucleo F401RE, Mass Storage, Enabled (generic 'Serial'), None, Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"
In file included from C:\Users\luism\Documents\Arduino\libraries\TeensyStep\src/RotateControlBase.h:5,

                 from C:\Users\luism\Documents\Arduino\libraries\TeensyStep\src/TeensyStep.h:3,

                 from C:\Users\luism\Documents\Arduino\DD_Stepper\DD_Stepper.ino:5:

C:\Users\luism\Documents\Arduino\libraries\TeensyStep\src/MotorControlBase.h: In instantiation of 'int TeensyStep::MotorControlBase<t>::getCurrentSpeed() const [with TimerField = TimerField]':

C:\Users\luism\Documents\Arduino\DD_Stepper\DD_Stepper.ino:275:44:   required from here

C:\Users\luism\Documents\Arduino\libraries\TeensyStep\src/MotorControlBase.h:92:44: error: passing 'const TimerField' as 'this' argument discards qualifiers [-fpermissive]

   92 |         return timerField.getStepFrequency();

      |                                            ^

In file included from C:\Users\luism\Documents\Arduino\libraries\TeensyStep\src/TeensyStep.h:26,

                 from C:\Users\luism\Documents\Arduino\DD_Stepper\DD_Stepper.ino:5:

C:\Users\luism\Documents\Arduino\libraries\TeensyStep\src/timer/stm32/TimerField.h:23:19: note:   in call to 'unsigned int TimerField::getStepFrequency()'

   23 |   inline unsigned getStepFrequency() { return stepTimer.getTimerClkFreq(); } //  clock of timer  or overflow?

      |                   ^~~~~~~~~~~~~~~~

exit status 1

Error compiling for board Nucleo-64.

Let me know if there is a work around otherwise I will have to measure the velocity from encoder.
Thanks

lmonteiro440, 2021-01-11:
The root cause is avoiding that rotate commands work as well.
HardwareTimer libs from STM32duino and TeensyStep at the moment not work well but no error at compiling process.
I´m using the last version.
Let me know if you need to do any test to solve it,
Thanks

lmonteiro440, 2021-01-21:
Since I did not receive any feedback here I took a time to understand the library and see what is wrong.
1 - There is a difference from the GCC compiler from Teensy (11 or newer) and  STM32duino in Platformio so as work around I declared it twice.
2 - the correct calculation for get the current speed on Timer is: inline unsigned getStepFrequency() const { return (stepTimer->getTimerClkFreq() / stepTimer->getPrescaleFactor()) / stepTimer->getOverflow(); } 
3 - I don´t know why you didn´t started the steptimer and I modified the rotateControlBase.h a little:
      void RotateControlBase<a, t>::doRotate(int N, float speedFactor)
    {
        if(this->isRunning())
        {
            this->err(mcErr::alrdyMoving);
            return;
        }

        //Calculate Bresenham parameters ----------------------------------------------------------------
        std::sort(this->motorList, this->motorList + N, Stepper::cmpVmax);
        this->leadMotor = this->motorList[0];

        if (this->leadMotor->vMax == 0)
            return;

        this->leadMotor->currentSpeed = 0;

        this->leadMotor->A = std::abs(this->leadMotor->vMax);
        for (int i = 1; i < N; i++)
        {
            this->motorList[i]->A = std::abs(this->motorList[i]->vMax);
            this->motorList[i]->B = 2 * this->motorList[i]->A - this->leadMotor->A;
        }
        uint32_t acceleration = (*std::min_element(this->motorList, this->motorList + N, Stepper::cmpAcc))->a; // use the lowest acceleration for the move
        uint32_t pullInSpeed;
                // Start moving----------------------------------------------------------------------------------------------
        isStopping = false;
        this->timerField.begin();
        accelerator.prepareRotation(this->leadMotor->current, this->leadMotor->vMax, acceleration, this->accUpdatePeriod, speedFactor);
        this->timerField.setStepFrequency(pullInSpeed);
        this->timerField.stepTimerStart();
        this->timerField.accTimerStart();
    }

Now everything is working fine, flawless with only 1 motor which is my case,  but let me know how I can help to keep your library up-to-date with this mods. 
I´ll modify and tuning the STM32 Timer and I can provide the final library. Let me know how.
I removed the automatically assignment of timers since I used several for others functions. I used 3, 9 and 11 since it´s enough for the task and are 16bits and with less channels. Let 32bits, advanced and 4 channels for other uses.
Keep me post.
Thanks

luni64, 2021-01-22:
Actually the STM32 code is from @ramboerik. I can't help you with any STM32 related issues because I currently don't use the STMs at all. However, I created a branch STM32. I'm willing to accept pull requests to this branch, but I need you STM32 folks to maintain them. Alternatively and maybe more practicable you can of course fork the repo and maintain your own "STM32 copy", I'm happy to put a link to it into the readme

ramboerik, 2021-01-23:
Hi @lmonteiro440. I didn't get any notifications for your issue but here is my late reply 😃 

> There is a difference from the GCC compiler from Teensy (11 or newer) and STM32duino in Platformio so as work around I declared it twice.  

I'm a bit unsure of this error and cannot reproduce it. Can you elaborate it?

> 2 - the correct calculation for get the current speed on Timer is: inline unsigned getStepFrequency() const { return (stepTimer->getTimerClkFreq() / stepTimer->getPrescaleFactor()) / stepTimer->getOverflow(); }
3 - I don´t know why you didn´t started the steptimer and I modified the rotateControlBase.h a little:

That part of the code has been quite untested but from your feedback I submitted a PR #95 to the STM32 branch.

> I removed the automatically assignment of timers since I used several for others functions. I used 3, 9 and 11

The current solution with `get_timer()` isn't optimal and more or less tuned for the stm32f429zi board. Other stm32f4 board don't have the same amount of timers and sometimes(as in your case) it's necessary to manually specify what timers to use. I would like to have a way of auto-detect what timers to use based on the current board and an option to manually specify what timers to use.

@luni64 About the maintenance: I don't know what's the best way to handle it atm. I like the idea of one repo that runs on multiple boards but at the same time it requires some effort from the maintainers to test new features across all boards.

lmonteiro440, 2021-01-23:
> Actually the STM32 code is from @ramboerik. I can't help you with any STM32 related issues because I currently don't use the STMs at all. However, I created a branch STM32. I'm willing to accept [...]

Don´t worry. @ramboerik is aware now and I will provide the information to him.

lmonteiro440, 2021-01-23:
> I'm a bit unsure of this error and cannot reproduce it. Can you elaborate it?
I had to modify the Steeper.cpp as following:
namespace TeensyStep
{
    constexpr int32_t Stepper::vMaxMax;
    constexpr uint32_t Stepper::aMax;

Using Platformio Arduino STM32duino F401RE

> > 2 - the correct calculation for get the current speed on Timer is: inline unsigned getStepFrequency() const { return (stepTimer->getTimerClkFreq() / stepTimer->getPrescaleFactor()) / stepTimer->getOverflow(); }
It worked flawless compared with oscilloscope value. In fact PreScale and Overflow needs to be sum of 1 but the difference is so small that I removed from formula,
> The current solution with get_timer() isn't optimal and more or less tuned for the stm32f429zi board. Other stm32f4 board don't have the same amount of timers and sometimes(as in your case) it's [...]
On the Hardware Timer Library of STM32duino getTimerClkFreq() returns exactly the oscillator clock I do not understand your your point. It returns 84000000 in my case (F401RE).
The rotateControlBase.h isn't started the stepper timer I don´t know how it was working in Teensy as well.
I set vpullin value inside the start of rotation as well.
        this->timerField.begin();
        accelerator.prepareRotation(this->leadMotor->current, this->leadMotor->vMax, acceleration, this->accUpdatePeriod, speedFactor);
        this->timerField.setStepFrequency(this->leadMotor->vPullIn);
        this->timerField.stepTimerStart();
        this->timerField.accTimerStart();
I will test it in a very fast solution where it will be modifying speed and stop it to invert direction pretty fast and I will look for overrideAcceleration as well and if it is ok I will post the results and final (for me) version of mods here.
If fact in a variant definition (PeripheralPins.c) for the used board and read what timers are available. The problem is that like my case I want to keep it tied since I use timer to others routines like read encoders and so on. If Teensy starts first it would use one or more timers that it could not. So I fix the 3 timers since I will use one kind of controller as well one motor. I have concerning how useful it would be.

ramboerik, 2021-01-23:
@lmonteiro440 What branch of TeensyStep are you using?  Can you post the content of your `platformio.ini`?

lmonteiro440, 2021-01-23:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nucleo_f401re]
platform = ststm32
board = nucleo_f401re
framework = arduino
lib_deps = luni64/TeensyStep@^2.1
check_tool = cppcheck
monitor_speed = 115200
monitor_port = COM[4]
build_flags = -Wl,-u,_printf_float,-u,_scanf_float

I cloned the github since the version that arduino/platformio find do not have STM32.

lmonteiro440, 2021-01-23:
I got the same idea from step frequency to get the stored values for update and pulse time.

   StepFreq = controller.getCurrentSpeed();
   AccTime = 1000000 / controller.getCurrentAcc();
   PulseTime = 1000000 / controller.getCurrentPulse();

Worked fine. I did it for debug reason but if some one think that it is useful....

ramboerik, 2021-01-24:
It's possible to specify git repositories directly in the `platformio.ini` to get the latest version of libraries that aren't published on pio's library page.

I used the following `platformio.ini` configuration and got no compiler error after modifying what timers to use.

```ini
[env:nucleo_f401re]
platform = ststm32
board = nucleo_f401re
framework = arduino
lib_deps = https://github.com/luni64/TeensyStep.git#STM32
check_tool = cppcheck
monitor_speed = 115200
monitor_port = COM[4]
build_flags = -Wl,-u,_printf_float,-u,_scanf_float
```

The timer modifications: 
```diff
--- .pio/libdeps/nucleo_f401re/TeensyStep/src/timer/stm32/TimerField.cpp.old   2021-01-24 07:52:03.973827900 +0100
+++ .pio/libdeps/nucleo_f401re/TeensyStep/src/timer/stm32/TimerField.cpp       2021-01-24 07:52:07.289824300 +0100
@@ -1,6 +1,6 @@
 #if defined(STM32F4xx)
 #include "TimerField.h"
 int TimerField::instances = 0;
-TIM_TypeDef* TimerField::timer_mapping[MAX_TIMERS] = { TIM1, TIM2, TIM4, TIM5, TIM6, TIM7, TIM8, TIM9, TIM10, TIM11, TIM12, TIM14 };
+TIM_TypeDef* TimerField::timer_mapping[MAX_TIMERS] = { TIM1, TIM2, TIM4, TIM5, TIM9, TIM10, TIM11 };
 // TIM3 and TIM13 used by HAL/FreeRTOS? doesn't work well to use.
 #endif
```
```diff
--- .pio/libdeps/nucleo_f401re/TeensyStep/src/timer/stm32/TimerField.h.old     2021-01-24 07:55:02.114852700 +0100
+++ .pio/libdeps/nucleo_f401re/TeensyStep/src/timer/stm32/TimerField.h  2021-01-24 07:45:04.022034900 +0100
@@ -7,7 +7,7 @@
 #include "../TF_Handler.h"

-#define MAX_TIMERS 12
+#define MAX_TIMERS 7
 class TimerField
 {
 public:
```

Please try the above pio config and report back if it still gives compiler errors 😃

lmonteiro440, 2021-01-24:
I know that I can point to here to load the library but I prefer customize and keep it locally.
Regarding the timers these modifications were the first thing that I did but TIM1 2 and 4 are been used by other tasks so I decided say goodbye to timerfield.cpp and did everything in the .h and fix it to Stepper to TIM 5, Acc to TIM9 and Pulse to TIM 11. Safer for me.

ramboerik, 2021-01-24:
I suspect that some local change in your TeensyStep files generates the GCC error as I'm unable to reproduce it  with a clean pio project. My question was if a clean project with TeensyStep generates the same error for you or if it's only present in your modified project. I'm currently unable to fix it as I cannot reproduce it.

> 1 - There is a difference from the GCC compiler from Teensy (11 or newer) and STM32duino in Platformio so as work around I declared it twice.
2 - the correct calculation for get the current speed on Timer is: inline unsigned getStepFrequency() const { return (stepTimer->getTimerClkFreq() / stepTimer->getPrescaleFactor()) / stepTimer->getOverflow(); }
3 - I don´t know why you didn´t started the steptimer and I modified the rotateControlBase.h

Summary so far:
1. Unable to reproduce
2. Fixed in branch STM32
3. Fixed in branch STM32

closed 2021-09-28 by luni64