luni64/TeensyStep #92: When no steps are needed, moveAsync() with stop() gets stuck?

LegacyMecha, 2021-01-13:
On a Teensy 3.2, I'm running a 3 motors and it seems (can't find the exact pattern) that if I do a moveSync() on them and they are all set to targetAbs with the exact same number as their current position (basically no steps are needed to move), and then I do a stop() immediately after, the output on the pins stop, getCurrentSpeed() = 100 and isRunning() = true. Can anyone confirm this weird behavior?
(Sorry, I can't post code for internal IP purposes).

LegacyMecha, 2021-01-13:
To fix this I have to check if there is anything to move before hand, and if so it runs fine. Can it be possible if the lead motor is zero this happens?

luni64, 2021-01-14:
Weird, I'll have a look at it on the weekend.

LegacyMecha, 2021-02-25:
Still can't figure this out. Is this an issue?

LegacyMecha, 2021-02-26:
I think it might have to do with "isRunning()" . isRunning() seems to be true even though "getCurrentSpeed()" is 0 in some cases. Any help with this would be great. Seems like when any motor with 0 travel is included like controller(M1,M2,M3), it'll hang on controller.stop() and Mx.isRunning() remains true even though it finishes.

luni64, 2021-02-26:
Can you please post a simple example which shows the behaviour?

Mike118, 2021-03-23:
Hello, I am not 100% sure but I think I noticed the same problem. 
Here is a small code to show the strange behavior: 

```
#include "TeensyStep.h"

#define NBSTEPPERS 2
const uint8_t DIRS[NBSTEPPERS] = {33, 31};
const uint8_t STEPS[NBSTEPPERS] = {20, 14};
Stepper stepperMotors[NBSTEPPERS] = {Stepper (STEPS[0], DIRS[0]),  Stepper (STEPS[1], DIRS[1])};
StepControl controller[NBSTEPPERS];

void initSteppers() {
  for (uint8_t i = 0; i < NBSTEPPERS; i++) {
    pinMode(DIRS[i], OUTPUT);
    digitalWrite(DIRS[i], LOW);
    pinMode(STEPS[i], OUTPUT);
    digitalWrite(STEPS[i], LOW);
    stepperMotors[i].setMaxSpeed(2000)       // steps/s
    .setAcceleration(2000); // steps/s^2
    if (!(controller[i].isOk())) {
      Serial.print( "ERROR : Stepper controller");
      Serial.print( i );
      Serial.println(" is not init");
    }
  }
}

int32_t getSpeed(uint8_t moteur) {
  return controller[moteur].getCurrentSpeed();
}

void checkStepperController() {
  for (uint8_t i = 0; i < NBSTEPPERS; i++) {
    if (!controller[i].isRunning()) {
      //Serial.println(" Not running ");
    } else {
      Serial.print(" Running ");
      uint32_t motorSpeed = getSpeed(i);
      Serial.println(motorSpeed);
    }
  }
}

void setup()
{
  Serial.begin(9600);
  initSteppers();
}

void loop()
{
  if(Serial.available()) {
    char c = Serial.read();
    if(c == 'g') {
      stepperMotors[0].setTargetRel(20000); 
      controller[0].moveAsync(stepperMotors[0]);
    } else if (c == 's') {
      controller[0].stopAsync();
    }
  }
 checkStepperController();
 Serial.print(" Position ");
 Serial.println(stepperMotors[0].getPosition());
}
```
If you put both g and s characters in the serial monitor at the same time you are stuck in a state where controller[i].isRunning() == true  and motorSpeed == 100 and the motor is not moving ...  ( Position do not increase )...  if you put g alone, it get unstuck.  But if you try few times you can get it stuck in a state where  controller[i].isRunning() == true  and motorSpeed == 100 stuck to 100 instead of going up to maxspeed  and the motor "is moving" ( or at least shown position increase ... )

I tested it with a teensy 3.2 ( you don't need anything attached on the teensy to test the behavior)

luni64, 2021-03-23:
Thanks for the test code. I'll have a look

Mike118, 2021-05-04:
Hello, did you take some time to look at the problem?
If I can do anything to help do not hesitate to ask for it...

luni64, 2021-05-05:
Frankly, I was distracted and forgot about it....

I had a look now and think I found the issue. Can you try if adding the following line 150 in `StepControlBase.h` fixes it for you? 

[image]

Additional remark (not the reason for the issue): Your test code is a bit rude since it is querying the state and position in a very tight loop.  The lib should be able to withstand this but it is always good to allow some milliseconds between those queries. 

Here some friendlier test code: 
```c++
#include "TeensyStep.h"

constexpr size_t NBSTEPPERS       = 2;           // no need for #defines in c++
const uint8_t DIRS[NBSTEPPERS]    = {33, 31};
const uint8_t STEPS[NBSTEPPERS]   = {20, 14};
Stepper stepperMotors[NBSTEPPERS] = {Stepper(STEPS[0], DIRS[0]), Stepper(STEPS[1], DIRS[1])};
StepControl controller[NBSTEPPERS];

void initSteppers()
{
    // for (uint8_t i = 0; i < NBSTEPPERS; i++)
    // {
    //     pinMode(DIRS[i], OUTPUT);  // no need for the pinModes, they are handled by TeensyStep
    //     digitalWrite(DIRS[i], LOW);
    //     pinMode(STEPS[i], OUTPUT);
    //     digitalWrite(STEPS[i], LOW);

    //     stepperMotors[i]
    //         .setMaxSpeed(2000)       // steps/s
    //         .setAcceleration(2000);  // steps/s^2

    //     if (!(controller[i].isOk())) // this is not implemented, will always return false
    //     {
    //         Serial.print("ERROR : Stepper controller");
    //         Serial.print(i);
    //         Serial.println(" is not init");
    //     }
    // }

    // c++ alternative
    for(Stepper& stepper : stepperMotors)
    {
        stepper
            .setMaxSpeed(2000)
            .setAcceleration(2000);
    }
}

int32_t getSpeed(uint8_t moteur)
{
    return controller[moteur].getCurrentSpeed();
}

void doCheck()
{
    static bool oldRunning  = false;
    static int32_t oldPos   = INT32_MAX;
    static int32_t oldSpeed = 0;
    bool running            = controller[0].isRunning();
    int32_t pos             = stepperMotors[0].getPosition();
    int32_t speed           = controller[0].getCurrentSpeed();

    if (oldRunning != running || oldPos != pos || oldSpeed != speed)
    {
        Serial.printf("Pos:%6d, Speed:%6d state: %s\n", pos, speed, running ? "running" : "stopped");
        oldPos     = pos;
        oldRunning = running;
        oldSpeed   = speed;
    }
}

IntervalTimer t;

void setup()
{
    while (!Serial) {}          // need to wait for serial is ready, otherwise you'll miss all Serial output for about 600ms
    initSteppers();

    t.begin(doCheck, 100'000);  // check in the background, every 100ms instead of polling in a thight loop

    Serial.println("start ---------------------------------");
    stepperMotors[0].setTargetRel(20000);
    controller[0].moveAsync(stepperMotors[0]);
    delay(1500);                                                // setting this to 0 generates the error in the original lib.
    Serial.println("stop ----------------------------------");
    controller[0].stopAsync();
    doCheck();
}

void loop()
{
}
```

For a delay of 1500ms this prints the expected result: 

[image]

For delay(0)  you'll get this
[image]
I.e. the controller stays in the running state. 

After the fix and still delay(0) you get the correct result
[image]

Let me know if this works for you

Mike118, 2021-05-12:
Thanks for this reply very helpful!  I will need time to fully test it and check if it solves all the problems I have noticed. ( It can be the exact same problem appearing in different cases )

Small detail about the modification suggested, what do you think about : 

if(newTarget == 0) this->timerField.end();
Instead of :
if(this->leadMotor->target == this->leadMotor->current) this->timerField.end();

And about the c++ alternative, I needed the iterator because even if it is not shown here, all my steppers have different accel and maxspeed that I stored in an array... So can I just create the iterator and increment it myself and be assured that all will be always be done 0 first then 1 etc ... ? 

Have a good day best regards.

luni64, 2021-05-12:
>Thanks for this reply very helpful! I will need time to fully test it and check if it solves all the problems I have noticed. ( It can be the exact same problem appearing in different cases )

Let me know if you find issues with the new code. If not I'd implement it in the next release.

>Small detail about the modification suggested, what do you think about :
>
>if(newTarget == 0) this->timerField.end();
Instead of :
if(this->leadMotor->target == this->leadMotor->current) this->timerField.end();`

Checking for 0 wouldn't catch the case when one accidentally calls `stopAsync` exactly when the motor already reached the target. 

>And about the c++ alternative, I needed the iterator because even if it is not shown here, all my steppers have different accel and maxspeed that I stored in an array... So can I just create the iterator and increment it myself and be assured that all will be always be done 0 first then 1 etc ... ?

Yes, you can be sure that it will iterate in the correct sequence. But, if you need the iterator its more straight forward to just use a for loop of course.

Mike118, 2021-05-22:
I think you solved the problem... For now, I did not manage to reproduce the problem I encountered many times in the past ...

About the case "  when one accidentally calls stopAsync exactly when the motor already reached the target. " I don't know what will be the exact difference between "newTarget == 0" and  "this->leadMotor->target == this->leadMotor->current "

if you have : A = B + C 
The two following operations "if(A == B)" and "if(C == 0)" should always return the same result ... 
Can the execution of the stop Async function be interrupted by another function modifying those values? 

Anyway, thank you for your great job.

luni64, 2021-05-22:
> don't know what will be the exact difference between "newTarget == 0" and "this->leadMotor->target == this->leadMotor->current "

You'll run into the issue if you stop the motor when it doesn't need to do any steps. This is the case when you didn't set a Target (the issue you observed). The same happens if the motor was already stopping and reached the target when you requested it to stop. -> current == target != 0.  Anyway, I didn't spend too much time in the analysis so I might be wrong :-)

Mike118, 2021-07-30:
Hello, 

I think that your modification reduced a lot the occurring of the problem but unfortunately, I fear that the problem is not fully solved. I still run into a strange case where " controller[i].isRunning() " return true but motor is not rotating and current position is not changing anymore ... By the way in this particular case controller.getCurrentSpeed() return a value higher or equal to 100 that stay constant. 
I suppose that this happens after using stopAsync ... It seems that the motor stop well but timerField is not ended correctly. 
I don't manage to make a simple repeatable example ... I run the same code on more than 15 teensy 3.2 each handling two motors ... And it appears from time to time on one of the teensy ...
During my tests I can be stopping a motor that was near is target or that was just starting and even both...  just starting and almost already on target and getting a stopAsync order ... 

I hope to finally solve this one day...

luni64, 2021-07-31:
Difficult to fix if it can not be reproduced. I can try to do some automated statistical movements to provoke the issue.

Mike118, 2021-08-02:
Yes difficult to solve in these condition I know ...
My last updates : I don't know if it can help but I modified the lib to be able te read the private attribute " target" position of the motor stepper and display it when the bug occur. When it occurs we have CurrentPost != TargetPos and  getspeed( ) > 0  (like 100 )  and all staying constant with controller.isRunning returning true... 
The good point is that in such case if you call stopAsync again, all seems to work again ... So Currently it seems that I managed to patch the consequences of the problem by calling stopAsyn( ) more time than I should ...

kolais, 2021-09-28:
Hi, just stumpled upon this issue which looks resolved for us after applying the @luni64 's patch from this thread. Is there any chance to get this valuable fix into the library (and deal with any further issues separately)?

luni64, 2021-09-28:
Thanks for the remainder, fixed,

kolais, 2021-09-28:
Cheers mate!

PR #113 (2021-09-28): Optimize speed of motor group, fix stopping when no steps left
fixes #92 
fixes  #107
closed 2021-09-28 by luni64