luni64/TeensyStep #94: setPosition while moving?

ZackM032488, 2021-01-21:
Hello,
Is it possible to set the current position on the fly?  I have a rotary application where a home switch will naturally be engaged once every rotation.  I'd like to be able to use this to my advantage by resetting the current position to 0 whenever the switch is engaged.  I have a workaround in place already, so it isn't critical by any means, but I think it would simplify what I'm doing a little bit.  Either way, thanks for the awesome library.

luni64, 2021-01-21:
I never did and never though of it, but it might work for the rotate controller (the step controller will definitely get confused if you change the position while it moves). Acceleration / deceleration might be problematic but its worth a try. 

Anyway, it is quite unusual to do such things. May I ask what you want to achieve?

ZackM032488, 2021-01-21:
The main goal is to have a self-homing and self-correcting system that manages to stay indexed during normal operation without the need for a dedicated homing cycle.  The application is for somewhat publicly accessible signage.  If someone accidentally moves the motor out of position I want it to simply fix itself over time.  Which I can do currently using the getPosition function when the switch is engaged and then again when it reaches its final position.  But I'd prefer to just set the position on the fly.

luni64, 2021-01-21:
Interesting. Sounds like some art installation? 

Maybe I miss something but instead of resetting the position counter when the index switch is hit you can simply readout and store the current motor position in the index ISR.  Whenever you need the current motor position subtract this stored value from getPosition(). The result should be the same as you would get by resetting the step counter in the index ISR.

ZackM032488, 2021-01-21:
Yeah, it's definitely art-adjacent.  I'll post it on the PJRC boards once it's public.

What you've describes sounds like essentially what I'm doing, though I'm not sure what you mean by "store the motor position in the index ISR."  I'm just using getPosition when the switch is hit and storing that in a local variable.

luni64, 2021-01-22:
>What you've describes sounds like essentially what I'm doing, though I'm not sure what you mean by "store the motor position in the index ISR." I'm just using getPosition when the switch is hit and storing that in a local variable.

You are probably doing exactly what I proposed :-). Just for completenes here some code showing how I would start (might need some debouncing if the switch is mechanical). If you are not familiar with the used lambda notation in `attachInterrupt()` you can of course pass in a dedicated function as callback. 

`m1`: some stepper object
`indexSwitchPin`: pin where the index switch is connected

```c++
int32_t offset = 0;

int32_t getStepperPos()
{
    return m1.getPosition() - offset;
}

void setup()
{
     //attach the index switch to an interrupt.  In the service routine just store the current position into 'offset'
     attachInterrupt(indexSwitchPin, []{ offset = m1.getPosition(); }, RISING);                                                                              
}
//....
```

ZackM032488, 2021-01-22:
Perfect.  Thanks for your help.

closed 2021-01-22 by ZackM032488