luni64/TeensyStep #96: can't compile..

patrixchee, 2021-01-29:
Dear Sir
Please Help
arduino IDE 1.8.13
connected to teensy 4.1
Lib.. TeensyStep  Ver. 2.1
testing on HelloStepper example file..
compile error..
In file included from C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_290002\HelloStepper.ino:19:0:
C:\Users\User\Documents\Arduino\libraries\TeensyStep\src/TeensyStep.h:39:72: error: 'TimerField' was not declared in this scope
 using RotateControl = TeensyStep::RotateControlBase<LinRotAccelerator, TimerField>;
                                                                        ^
C:\Users\User\Documents\Arduino\libraries\TeensyStep\src/TeensyStep.h:39:82: error: template argument 2 is invalid
 using RotateControl = TeensyStep::RotateControlBase<LinRotAccelerator, TimerField>;
                                                                                  ^
C:\Users\User\Documents\Arduino\libraries\TeensyStep\src/TeensyStep.h:40:69: error: 'TimerField' was not declared in this scope
 using StepControl = TeensyStep::StepControlBase<LinStepAccelerator, TimerField>;
                                                                     ^
C:\Users\User\Documents\Arduino\libraries\TeensyStep\src/TeensyStep.h:40:79: error: template argument 2 is invalid
 using StepControl = TeensyStep::StepControlBase<LinStepAccelerator, TimerField>;
                                                                               ^
HelloStepper:22: error: 'StepControl' does not name a type
 StepControl controller;    // Use default settings 
 ^
HelloStepper: In function 'void loop()':
HelloStepper:31: error: 'controller' was not declared in this scope
   controller.move(motor);    // Do the move
   ^
'StepControl' does not name a type

Many Thanks...

luni64, 2021-01-29:
TeensyStep is currently not compatible to the T4.x boards.

patrixchee, 2021-01-29:
ok,,,,thanks
will try on teensy 3.6.
many thanks
best regards....

closed 2021-01-29 by luni64

miracuruzrh, 2021-04-23:
Unfortunately I have only a teensy 4.1 is there a chance that this library will be updated to have it also for 4.1?

luni64, 2021-04-23:
Yes, I'm currently working on it

Nickperillo, 2021-05-17:
Any news on updating this library for the Teensy 4 boards? Looks Like a great library but I only have the newer boards.

luni64, 2021-05-17:
Since, technically this is not an issue but  pops up regularly, I opened a corresponding thread in the discussions section of the repo https://github.com/luni64/TeensyStep/discussions/105#discussion-3369482
Might be easier to discuss there.