luni64/TeensyStep #97: Control 25 Stepper motor independent

nuthapongl, 2021-02-21:
Hi luni64,
I have project to control 25 pieces of 28byj-48 stepper motor independent . Is it possible to control it ? and If it's work what is function can I use. Thanks :)
my project is similar this video https://www.youtube.com/watch?v=0o_9CHYeRvI

mfkenson, 2021-02-25:
lets get a teensy and try! according to the doc its up to 10 motors in sync mode.
However getting 2 more teensy boards are affordable. I have just lighted mine up (2 motors) and everything work as expected. I really enjoy the simplicity using this lib.

ramboerik, 2021-02-25:
Hi @nuthapongl! 

What kind of stepper driver are you going to use? This library is designed for stepper drivers with two inputs, STEP and DIR. Simpler driver like ULN2003  which requires four pins won't work.

ramboerik, 2021-02-25:
@mfkenson The library has support for 4 groups of 10 steppers each. The '4 groups' is a hardware limit but 10 steppers per group is not. It's possible to allow more than 10 steppers per group by editing the library. One Teensy may control more than 40 steppers if running them in sync. 
The question is how good the result turns out if running them in sync.

luni64, 2021-02-25:
Sorry, I somehow missed your post.  Everything @ramboerik  wrote above is correct of course. Looking at the linked video I'm not sure if TeensyStep is the right tool here. You probably want to change the mirror position whenever something changes. With TeensyStep you'd have to wait until the current movement is finished and then start a new one. 

Regarding the drivers: The 28byj-48 are voltage controlled and unipolar . To drive them with a standard bipolar stepper driver you need to manipulate the motors. This is possible but tedious.  See e.g. here https://www.youtube.com/watch?v=ROFjf_7IrRA

You are probably better of with some simple lib like "accelStep" or "stepper".

mfkenson, 2021-02-25:
@ramboerik Thanks! good to know that. I am working on a homemade 6-axis robot arm (i.e. 6 steppers). Trying to heavily rely on TeensyStep for the firmware. What you said sounds very positive.

@luni64 However I am little bit confused about "wait until the current movement". My design is let my laptop manage trajectory generation and keep sending the target joint coordinates via USB serial to teensy for execution (using `setTargetAbs`, `moveAsync`).

If the teensy has to wait until current movement, then I have to find another way to make the trajectory smooth enough. Maybe modifying the Path_Following/RoseFunctionFollower.ino example..?

ramboerik, 2021-02-25:
@mfkenson You won't be pleased by the result as your machine will stop at every single point your laptop generates, even if they are at the same line as the library can't do on the fly target updates. The steppers will accelerate  and deaccelerate for **every** point.   
 I did the very same thing and ended up writing a motion planner for TeensyStep here: https://github.com/ramboerik/TeensyStep/tree/feature-point-motion.  The code is however only tested for stm32, not well documented how to use and currently only for 3 axis but could quite easy be extended. 

My branch allows multiple points to be added to the steppers and it calculates the max speed for all axis when visiting the given points.

luni64, 2021-02-25:
>If the teensy has to wait until current movement, then I have to find another way to make the trajectory smooth enough. Maybe modifying the Path_Following/RoseFunctionFollower.ino example..?

Yes, basically this just uses a PID and override speed to have the motor following a moving target. But then you can only have 4 motors per Teensy.  

The motion planner from @ramboerik  sounds more promising I'd say.

mfkenson, 2021-02-25:
Thank you @ramboerik @luni64 for everything.

I just quickly look into the code. Basically the motion planner is to compute the path (as list of targets over time) before calling the move function. Awesome!  Let me see what I could do with this. It helps a lot anyway.

ramboerik, 2021-02-25:
No problems 😄 Feel free to ask about it!

mfkenson, 2021-02-26:
@ramboerik I can see the beauty of calculate and setting the "Targets" using your MotionPlanner to motorX and motorY. I believe I can get start my mod from there. Thanks!

However I encountered a problem when playing around the motion planner branch.
Even a very basic example with my teensy 3.5 would die when calling the moveAsync function.

I found out where it died.

inside doMove function in StepControlBase.h
`this->timerField.setStepFrequency`
https://github.com/ramboerik/TeensyStep/blob/f13f565022d8e39db4be7471fd86f9860f507fde/src/StepControlBase.h#L144

Following the rabbit:

The teensy is then dead until reset after calling setThisReload
`stepTimer.setThisReload(F_BUS / f);`
https://github.com/ramboerik/TeensyStep/blob/f13f565022d8e39db4be7471fd86f9860f507fde/src/timer/teensy3/TimerField2.h#L127

Let me keep debugging. So far I can see the difference between your motion planner branch and here the origin is the order of making `this->timerField.begin();`, inside doMove function in StepControlBase.h.

mfkenson, 2021-02-26:
I suddenly realized that you said the code is tested for stm32 only. Let me try to compare the difference between them!
 
https://github.com/ramboerik/TeensyStep/blob/f13f565022d8e39db4be7471fd86f9860f507fde/src/timer/stm32/TimerField.h

vs
 https://github.com/ramboerik/TeensyStep/blob/f13f565022d8e39db4be7471fd86f9860f507fde/src/timer/teensy3/TimerField2.h

mfkenson, 2021-02-27:
More update on teensy 3.5:

I put the line 
https://github.com/ramboerik/TeensyStep/blob/f13f565022d8e39db4be7471fd86f9860f507fde/src/StepControlBase.h#L144
```
this->timerField.setStepFrequency
```
below 
```
if(startTimers)
        {
            this->timerField.begin();
            this->timerField.stepTimerStart();
            this->timerField.accTimerStart();
        }
```
The teensy would not die there. And ISR doStep started to work.

However right now it die when calling 
`timerField.triggerDelay();` inside doStep. (if I delete this line then teensy does not die)

Will continue looking into timer stuff.

ramboerik, 2021-03-17:
@mfkenson I'm glad that you got it to work! Could you submit a PR to my repo so I can verify it for stm32 also :) ? Sometime in the future I would like to make the MotionPlanner generic and accept any given number of axis.

Just out of curiosity, how is your laptop sending the coordinates to the teensy and how often/when do you recalculate the path? My implementation builds on the fact that the whole path/trajectory is available when doing the path calculation and must always stop before calculating the next path.

and another question,
Have you noticed any stepper aliasing problem when running multiple steppers in sync? The Bresenham' algorithm TeensyStep uses might produce aliasing problem for the slave steppers. It varies with the steppers speed/distance but sometimes you can hear that the slave steppers don't run as smooth as the master stepper. 

The grbl CNC firmware have implemented a simple solution to the problem and they discuss it here: https://github.com/gnea/grbl/blob/master/grbl/stepper.c#L280

closed 2021-09-28 by luni64