luni64/TeensyStep #99: controller is not allocated

gandalf15, 2021-03-18:
Hello luni64,
firstly, I want to thank you for this library.  Secondly, I would like to ask you for  your help. I am trying to control two steppers independently with Teensy 3.6. Also I am using HW Serial1 and HW Serial2 (I do not know if HW serial has anything to do with timers). Therefore, I created two controllers and two motors. Only one was working. So, I tried to figure out where the problem is. When I call `controller.isOK()` it returns false and `controller.isAllocated()` returns false.  I do not know why. So, I tied a minimal code like this. Using latest library Teensystep 2.1 in PlatformIO. I also tried using ArduinoIDE. The result is the same.
```
#include "TeensyStep.h"

Stepper motor(12, 38);       // STEP pin: 2, DIR pin: 3
StepControl controller;    // Use default settings 

void setup()
{
}

void loop() 
{
    if (!controller.isOk()) {
      Serial.println("Controller is Not OK!");
    }
    if (!controller.isAllocated()) {
      Serial.println("Controller is Not allocated!");
    }
  delay(500);
}
```
The serial output is :
```
Controller is Not OK!
Controller is Not allocated!
```

luni64, 2021-03-19:
The `isOK` function is just a non functional left over from some tests and should have been removed long ago. `isAllocated()` returns true if the controller allocated a timer. Since timers are a very limited resource, the controller only allocates a timer when it actually moves motors and releases it afterwards. So, the false in your code makes sense. 

Can you post some code showing your original problem?

gandalf15, 2021-03-19:
Thank you @luni64 for the explanation. Unfortunately, I did not find any documentation for these methods on [the web page](https://luni64.github.io/TeensyStep/documentation/stepControl/). With the knowledge how the methods work, I was able to debug my code. Now, it all works as it should. The problem was in my code and your library works as expected.

closed 2021-03-19 by gandalf15