luni64/TeensyStep #102: Enable/Disable stepper causes lost position/steps

elfren, 2021-05-12:
Enabling and disabling the stepper causes either lost position or lost steps when using DM542T driver from StepperOnline (https://www.omc-stepperonline.com/download/DM542T.pdf).  I'm not seeing the same issue with DRV8825 (https://www.ti.com/lit/ds/symlink/drv8825.pdf?ts=1620734575016).   One difference between the two drivers is the timing requirements.  DM452T requires 5us before DIR and DRV8825 requires 650ns before STP.

Repro using modified example.  When the digitalWrite commands are commented out, the stepper moves the correct distance.

`/*==========================================================================
 * This is a minimal sketch showing the usage of TeensyStep
 *  
 * STEP Pulses on Pin 2    (can be any pin)
 * DIR  Signal on Pin 3   (can be any pin)
 * 
 * The target position is set to 1000 steps relative to the
 * current position. The move command of the controller 
 * moves the motor to the target position.  
 *  
 * Default parameters are 
 * Speed:          800 steps/s
 * Acceleration:  2500 steps/s^2
 * 
 * (slow, but good to start with since they will work on any normal stepper)
 *
 ===========================================================================*/

#include "TeensyStep.h"

Stepper motor(3, 2);       // STEP pin: 3, DIR pin: 2, ENA pin: 4
StepControl controller(3,5000);    // Pulse Width, Update period
int targetSteps = 1600; //  steps per revolution: 200, microsteps: 32  
 int state = 7;
void DoSteps()
{
  for (int i = 0;i < 4; i++) // Do 1 revolution
  {
      delay(100);
      Serial.print ("Pin4 for loop:" );
      state = digitalRead(4);
      Serial.println(state);
    
      digitalWrite(4,LOW); // Enable
      Serial.print ("Pin4 after digitalWrite Low:" );
      state = digitalRead(4);
      Serial.println(state);
      delay(6);
      motor.setTargetRel(targetSteps);
      controller.move(motor);    // Do the move
  
      delay(6);
      digitalWrite(4,HIGH); // Disable
  }

}

void setup()
{
  pinMode(4, OUTPUT); //Initialize pin 4
  digitalWrite(4,HIGH); // Disable
  motor.setAcceleration(10000);
  motor.setMaxSpeed(10000);

  DoSteps();
}

void loop() 
{

}`

luni64, 2021-05-12:
The behaviour of a disabled->enabled stepper driver obviously depends on the used driver. Some drivers just continue with the same current pattern after enabling, others might start from a "zero" pattern which would generate some step losses. Of course  the library can't change the behaviour of the driver. 

Anyway, I strongly recommend to not disable the driver if your application needs to keep the position while cycling the enable signal. The  Leadshine DM542 (the stepper online version might be an OEM version or a clone?) has a setting to reduce stand still current by 50% to reduce heat/energy consumption while ensuring to keep the stepper position. 

https://www.cnc-spezialist.de/images/PDF/DM542E_Engcnc.pdf

Hope that helps

elfren, 2021-05-12:
Unfortunately, there can be a lot of manual adjustments between moves, so the steppers need to be disabled but not necessarily after every move.  The lost position isn't reported by TeensyStep, so currently I don't know of any way to recover that.

closed 2021-05-12 by elfren