luni64/TeensyStep #110: A way to return current step count when using RotateControl?

joonasstenvall, 2021-08-31:
Hi, I need to drive a stepper motor with a speed set by a value(-1000 to 1000). And when I press a button, the current position will be set to a position array. I was able to do this with accelstepper library but that library is not accetable due to no acceleration support with multiple steppers. 

`void drivemotors()
{
	menu_system.change_screen(&runMotorsSetPositionsMenu_screen1);
	//Serial.println("drive");
	

	indexValue = indexValueLCD-1;

	///Serial.printf("poscount value %d \n", positionCount);
	trueLoop = true;

	while (trueLoop)
	{
		int oldspeed = speed;
		myTransfer.tick(); //This gets the serial data and button values

		if (inputArray[xbLB])
		{
			//Serial.printf("index value %d \n", indexValue);
			if (indexValue - 1 >= 0)
			{
				indexValue--;
				//Serial.printf("index value %d \n", indexValue);
				indexValueLCD = indexValue + 1;
				menu_system.update();
				moveToIndex(indexValue);
			}
			inputArray[xbLB] = 0;
			
		}
		else if (inputArray[xbRB])
		{
			//Serial.printf("index value %d \n", indexValue);
			if (indexValue + 1 < positionCount)
			{
				indexValue++;
				//Serial.printf("index value %d \n", indexValue);
				indexValueLCD = indexValue + 1;
				menu_system.update();
				moveToIndex(indexValue);
			}
			inputArray[xbRB] = 0;
		}

		lenghtStepper.setSpeed(inputArray[xbLSTICKAX]); //gamepad analog values
		turnStepper.setSpeed(inputArray[xbRSTICKAX]);	//gamepad analog values
		tiltStepper.setSpeed(inputArray[xbRSTICKAY]);	//gamepad analog values

		lenghtStepper.runSpeed();
		turnStepper.runSpeed();
		tiltStepper.runSpeed();
		
		if (oldspeed != speed)
		{
			menu_system.update();
		}

		if (inputArray[xbX])
		{
			//Serial.printf("index value %d \n", indexValue);
		
			positions[indexValue][0] = lenghtStepper.currentPosition();
			positions[indexValue][1] = turnStepper.currentPosition();
			positions[indexValue][2] = tiltStepper.currentPosition();
		
			inputArray[xbX] = 0; //Set the button value to not re trigger
			if (indexValue + 1 < positionCount)//try to make it so that does not increment if almost at last position
			{
				if (positions[indexValue + 1][0] == 0 || positions[indexValue + 1][2] == 0 || positions[indexValue + 1][2] == 0)
				{//Try to check if next value is zeroes maybe not correct
					indexValue++;
					indexValueLCD = indexValue + 1;
				}
			}
			//Serial.printf("index value %d \n", indexValue);
			menu_system.update();

		}

	}
	
	steppers.moveTo(zeroPos);
	steppers.runSpeedToPosition();

	//stepperLenght.setPosition(lenghtStepper.currentPosition());
	//stepperTurn.setPosition(turnStepper.currentPosition());
	//stepperTilt.setPosition(tiltStepper.currentPosition());
	gotoMain();
}
`
I tried to set the positions with my old code and to set the current positions to the teensysteps but that did not work. I theorise that the step counts are different because of the different acceleration implementations.

luni64, 2021-08-31:
Doesn't stepper.getPosition() work?

joonasstenvall, 2021-08-31:
Oh my, It does. I must have not used it correctly. My bad.

closed 2021-08-31 by joonasstenvall