luni64/TeensyStep #117: Help with driving motors with set speed.

joonasstenvall, 2021-11-06:
Hi, I was wondering if anyone could help me a bit. [https://github.com/luni64/TeensyStep/issues/57](Found this) thread which has helped me a bit but the code is a bit buggy and does not always change directions or start moving.

`void pid(RotateControl &controller, int32_t target)
{  
	constexpr unsigned PID_Interval = 20; // ms  
	constexpr float P = 0.01;// (P)roportional constant of the regulator needs to be adjusted (depends on speed and acceleration setting)

	static unsigned lastTick = 0;

  if (millis() - lastTick > PID_Interval)
  {   
	//float delta = (target - stepperPan.getPosition()) * (P / PID_Interval);
	float delta = target * (P / PID_Interval);  // This implements a simple P regulator (can be extended to a PID if necessary)
	float factor = std::max(-1.0f, std::min(1.0f, delta)); // limit to -1.0..1.0

    controller.overrideSpeed(factor); // set new speed

	lastTick = 0;
  }
}

void contRotation(RotateControl &rotator, int32_t input)
{
	int32_t deadzone = 50;

	Serial.printf("input %d ", input);

	if (input > deadzone || input < -deadzone)
	{
		Serial.printf("input 2 %d", input);
		pid(rotator, input);
	}else{
		pid(rotator, 0);
	}

	Serial.println();
	
}
`
How should I go about fixing this code?

Thanks!

luni64, 2021-11-06:
Can you post a compilable example (as small as possible, without additional hardware / libraries) which shows the problem?

joonasstenvall, 2021-11-06:
Hello, I could certainly do so but it would take me a bit. But would you happen to have rotator code that can be driven with a setspeed? What I would best imagine this is turning a video game character with the look stick on a controller. That's what I'm trying to do but for turning the stepper motors. Would not even have to be mm/s.

joonasstenvall, 2021-11-06:
Basically just drive the motors with analog stick

joonasstenvall, 2021-11-06:
`void pid(RotateControl &controller, int32_t target)
{  
	constexpr unsigned PID_Interval = 20; // ms  
	constexpr float P = 0.01;// (P)roportional constant of the regulator needs to be adjusted (depends on speed and acceleration setting)

	static unsigned lastTick = 0;

  if (millis() - lastTick > PID_Interval)
  {   
	//float delta = (target - stepperPan.getPosition()) * (P / PID_Interval);
	float delta = target * (P / PID_Interval);  // This implements a simple P regulator (can be extended to a PID if necessary)
	float factor = std::max(-1.0f, std::min(1.0f, delta)); // limit to -1.0..1.0

    controller.overrideSpeed(factor); // set new speed

	lastTick = 0;
  }
}

void contRotation(RotateControl &rotator, int32_t input)
{
	int32_t deadzone = 50;

	Serial.printf("input %d ", input);

	if (input > deadzone || input < -deadzone)
	{
		pid(rotator, input);
	}else{
		pid(rotator, 0);
	}

	Serial.println();
	
}

void drivemotors()
{
	trueLoop = true;

	stepperLenght
      .setMaxSpeed(10000)
      .setAcceleration(75000);
	
	RotateControl rotcot1;

  	rotcot1.rotateAsync(stepperLenght);
  	rotcot1.overrideSpeed(0); // start with stopped slide

	
	stepperPan
      .setMaxSpeed(10000)
      .setAcceleration(75000);
	
	RotateControl rotcot2;

  	rotcot2.rotateAsync(stepperPan);
  	rotcot2.overrideSpeed(0); // start with stopped slide

	stepperTilt
      .setMaxSpeed(10000)
      .setAcceleration(75000);
	
	RotateControl rotcot3;

  	rotcot3.rotateAsync(stepperTilt);
  	rotcot3.overrideSpeed(0); // start with stopped slide

	while (trueLoop)
	{
		myTransfer.tick(); //This gets the serial data and button values

		//Does not actuate sometimes
		contRotation(rotcot1,inputArray[xbLSTICKAX]/*input value from joystick*/);
		contRotation(rotcot2,inputArray[xbRSTICKAX]);
		contRotation(rotcot3,inputArray[xbRSTICKAY]);

	}

}
`
Here is the code you asked for.

luni64, 2021-11-06:
> Hello, I could certainly do so but it would take me a bit. But would you happen to have rotator code that can be driven with a setspeed? What I would best imagine this is turning a video game [...]

Your posts are a bit difficult to understand. What I understand is you simply want to do is set the speed of the motor with a pot or some other analog input?  If this is what you want I can post an example. 

If not: 
The code you posted is not complete. I don't want to be rude, but if you want me to help you I need  a small but COMPLETE  piece of code that I can copy to Arduino, compile and run it. The code you posted doesn't even have setup() and loop()... I also need a clear description of the problem you observe and how to reproduce it.

joonasstenvall, 2021-11-06:
> Your posts are a bit difficult to understand. What I understand is: you want to set the speed of the motor with a pot or some other analog input? If this is what you want I can post an example.
I'm sorry for this as I'm not as good at English as I would like.

> If not:
The code you posted is not complete. I don't want to be rude, but if you want me to help you I need a small but COMPLETE piece of code that I can copy to Arduino, compile and run it. The code you posted doesn't even have setup() and loop()... I also need a clear description of the problem you observe and how to reproduce it.
Oh my! I did not realise this. My apologies. 

I did end up fixing this problem of mine as I had copy pasted the intial max speed and acceleration values and that caused the motors to stall while producing next to no noise.

I hope that I have not wasted too much of your time.

Thank you for still trying to help me!

closed 2021-11-06 by joonasstenvall