luni64/TeensyStep #118: Motor stalling exactly at maxSpeed 65536.

harshdobariya, 2021-11-18:
I have tried many combinations of settings, drivers, power supply, accelerations, etc but **the motor just won't go above 65535steps/sec**. It works fine at up to 65536steps/sec but as soon as I increase like even by a 1steps/sec the motor just stalls (stops spinning and starts vibrating between single steps.) The motor also stalls at high microstepping but same steps/sec. That is even at low RPM it can’t take more than 65536steps/sec.

Tried different accelerations but same result. Stall at 65536steps/sec and above.

### Compiler used - Arduino1.8.13
### Microcontroller used – Teensy3.6

The combination of stepper motor, drivers and setting are listed in the table below:
(Power source used for the following testing is 24V 3A)

### Stepper motor used:
IMPULSE IM86HS78-5504X-07J
Resistance/Phase: 1.16ohm
Inductance/Phase:6.8mH

Stepper Online 23HS45-4204S
Resistance/Phase: 0.9ohm
Inductance/Phase: 3.8mH
Driver & Current Settings:

### Chinese Driver:
TB6600 (4.0A)
DM542 (4.2A)

### Microstepping:
3200
6400
12800

### Code is also the simplest, just checking the max speed of stepper motor:
`#include "TeensyStep.h"

Stepper motor(13, 11);
StepControl controller;

void setup() {
  Serial.begin(9600);
  
  motor.setMaxSpeed(66000);
  motor.setAcceleration(60000);  
}

void loop() {
  motor.setTargetRel(600000);
  controller.move(motor);
  delay(1000);

  motor.setTargetRel(-600000);
  controller.move(motor);
  delay(1000);
}`

Is this counter overflow problem? Or am I missing out on something?

luni64, 2021-11-19:
I tested the output of your code with an LA.  For testing I changed the target positions to 100'000 and -100'000. I didn't observe anything strange. Here a few traces:

Number of generated Pulses: 0K
[image]

Zoomed in at the beginning:
[image]

Zoomed in at the center: (average Frequency: 65.9kHz
[image]

Looks like the library does what it is supposed to do?

luni64, 2021-11-19:
I just remembered that I answered the very same question on stackoverflow a couple of month ago with the same result. Did you check anything I recommended to check there? If so, what was the result? 

https://electronics.stackexchange.com/questions/568159/whats-happening-to-my-stepper-motor-at-high-rpm

harshdobariya, 2021-11-20:
Hello @luni64, thank you for your time.
I know you have answered the same question on stackexchange. I have tried everything you said.

### Issue with the following:
**Driver or motor:**
I have a cheap Chinese router with ready-made controller. Those driver-motor combination can, with its own controller, easily go above 66kHz speed. When I tested those driver-motor combination with Teensy3.6 and TeensyStep2.1 library, I could able to spin the motor easily till max frequency of 65kHz but at 66kHz it stalls.

**Using 3.3V and not 5V:**
I have tried using 5V logic level convertor (MC74HC125N), but got the same result.

### Now I have tried looking at the pulse of the step pin produced by the following code:
I have used Visual Studio Code. (Earlier I was using Arduino1.8.13)
```
[env:teensy36]
platform = teensy
board = teensy36
framework = arduino
lib_deps = luni64/TeensyStep@^2.1
```

```
#include <Arduino.h>
#include <TeensyStep.h>

Stepper motor(9, 10);
StepControl controller;

uint8_t pin=8;

void setup(){  
  pinMode(pin, OUTPUT);
  motor.setMaxSpeed(65535);    //either 65535 or 65536
  motor.setAcceleration(100000);
}

void loop() {
    motor.setTargetRel(100000);

    digitalWrite(pin, HIGH);
    controller.move(motor);
    digitalWrite(pin, LOW);
    delay(100);
}
```

So I have used **HANTEK6022BE** for measuring the step frequency.
Attached pin 9 (step pin) to Channel1 Yellow
Attached pin 8 (status pin) to Channel2 Blue

### Here is the oscilloscope result when _setMaxSpeed = 65535:_
Both graphs upper and lower belongs to the same program parameters, just different division scale.
As you can see the step channel(yellow) shows deacceleration and acceleration at the ending and staring of the cycle. But...

[image]
.

### Here is the oscilloscope result when _setMaxSpeed = 65536:_
Again all the three graph belong to the same program parameters but different division scale.
In this graph you can see there is no phase of acceleration and deacceleration and the step frequency directly jumps to 65536Hz. Even at 2ms/DIV we can't see the pulse accelerating to the maxSpeed. 

[image]
.

This could be the reason for motor stalling at maxSpeed 65536 or above. But in the first place why is it happening?

luni64, 2021-11-21:
Ok, that is good information. I'll have a look later today

luni64, 2021-11-21:
I can reproduce that. Looks like it is only happening at those speeds. E.g 70kHz works normally. I'll have a closer look. Might take some time...

luni64, 2021-11-22:
Can you please try if the version in the development branch fixes your issue?

harshdobariya, 2021-11-24:
As I was out of town i couldn't reach sooner.
I will test the fix and let you know the result soon. As I am new to GitHub I don't know how to fix/update the particular file. Do I need to update my whole TeensyStep2.1 library?

luni64, 2021-11-24:
> As I am new to GitHub I don't know how to fix/update the particular file. Do I need to update my whole TeensyStep2.1 library?

Yes, best to completely replace the library.  First switch to the development branch with  
the branch button and select `develop` from the dropdown list. 

[image]

Then use the code button and select `Download zip`. 

[image]

In the long run I strongly recommend to learn 'some git'. Especially how to _clone_ a repository and _pull changes_ .

harshdobariya, 2021-11-24:
So I have updated the fix and checked the signals with the same program and parameters on the oscilloscope and we got the accelerating and decelerating signals.
I have also tested one stepper motor with the same program and it started going more than 65535Hz frequency.
Thanks for your time and immediate response.
Also I'll learn Git.

PR #120 (2021-11-25): Fixes an issue with uint32_t overflow for large speed and a restarting problem for overrideSpeed(0)
fixes #118 and #119
closed 2021-11-24 by harshdobariya