luni64/TeensyStep #119: OverrideSpeed(0) Locks Up/ Wont Restart motor

Atomfusion1, 2021-11-20:
If you set overrideSpeed() from a speed down to exactly 0 
overrideSpeed does not restart pump when you change it to any other value 
Included Code 
type: 
A20 // motor starts moving
A10 // motor slows but keeps moving 
A0 // motor stops
A20 // Motor does not move 
etc. 

`#include "TeensyStep.h"

RotateControl controller1;
Stepper motor_1(1, 2);  //STEP pin =  1, DIR pin = 2

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);                 // Serial Port Baud Rate Communication with Labview
  unsigned long start = millis();
  while (!Serial) {
    if (millis() - start > 200)
      break;
  }
  Serial.println(F("Startup"));
  // Motor Setup
  motor_1
  .setMaxSpeed(100)       // steps/s
  .setAcceleration(200000); // steps/s^2
  motor_1.setMaxSpeed(20000);
  motor_1.setAcceleration(200000);
  // controller setup
  controller1.rotateAsync(motor_1);
  controller1.overrideSpeed(0.0f);
}

void loop() {
  // put your main code here, to run repeatedly:
  if (Serial.available() > 0) SerialCase(SerialRead());    // Check If Serial Data available and deal with it
}

String SerialRead (void)                // Serial Read Function ..
{
  //  TIMSK4 |= (0 << OCIE4A);  // disable timer compare interrupt
  delay(1);
  String Temp = "";
  while (Serial.available() > 0)
  {
    Temp += (char) Serial.read();       // Get Serial Data
    delay(2);
  }
  return Temp;                          // Return Data
}

void SerialCase (String input) // Switch Case Control
{
  switch (input[0])
  {
    case 'A':                                                                        // Set Pump 1 Speed 0 to 100%
      {
        String tempString;
        for (int i = 1; i <= 3; i++) {
          tempString += input[i];
          if (input[i] == '\n') i = 3 + 1;
        }
        // convert the string to integer
        unsigned long inttemp = 0;
        inttemp = tempString.toInt();
        float rpm = inttemp;
        controller1.overrideSpeed(rpm / 200.0f);
        controller1.rotateAsync(motor_1);
        Serial.println(rpm / 200.0f);
      }
      break;
  }
}`

luni64, 2021-11-21:
There was a subtle bug in TeensyStep which caused the issue. The code in the develop branch should fix this. Can you please test it? 

I also had a look at your code which seems a bit over engineered. Here a simpler proposal doing the same: 
```c++
#include "Arduino.h"
#include "TeensyStep.h"

void SerialCase();

RotateControl controller1;
Stepper motor_1(1, 2); //STEP pin =  1, DIR pin = 2

void setup()
{
    while (!Serial && millis() < 1000) {}

    Serial.println(F("Startup"));

    // Motor Setup
    motor_1
        .setMaxSpeed(20'000)       // steps/s
        .setAcceleration(200'000); // steps/s^2

    // controller setup
    controller1.rotateAsync(motor_1);
    controller1.overrideSpeed(0.0f);
}

void loop()
{
    // put your main code here, to run repeatedly:
    if (Serial.available() > 0) SerialCase(); // Check If Serial Data available and deal with it
}

void SerialCase() // Switch Case Control
{
    switch (Serial.read())  // read one char from Serial
    {
        case 'A': // Set Pump 1 Speed 0 to 100%
        {
            int rpm = Serial.readStringUntil('\n').toInt();
            controller1.overrideSpeed(rpm / 200.0f);
            //controller1.rotateAsync(motor_1); <============ NEVER call move/rotate if the motor is already moving/rotating!!!
            //Serial.println(rpm / 200.0f);
            break;
        }
        case 'B':
            // ....
            break;

        default:
            Serial.println("Unknown Command");
    }
}
```

Atomfusion1, 2021-11-21:
Well that re-write taught me a few tricks lol Thanks.. I have been having issues with that override(0) for some time.. been adding .001 to everything in order for it not to lock up .. the Dev. branch works perfectly. 

I know you wrote the code to synchronize pumps etc. but I think a few of us in the forums are trying to use your code for 4 independent controllable pumps, for instance I am trying to use it to control a stepper peristaltic pump while feeding it 50-100 different speeds a second, this allows for mimicking of a heartbeat or other abnormal flow 

I know somewhere in your code I have read not to spam overrideSpeed(), but even at 1-2ms dt I have had no issues other than the lockup with (0). maybe add the above code as a simple example of SpeedControl 

Thanks for the quick reply and your time

luni64, 2021-11-21:
Great, I'll wait a week or so before I pull it into the main branch. Let me know if the change has some unwanted side effects. 

> I know somewhere in your code I have read not to spam overrideSpeed(), but even at 1-2ms dt I have had no issues other than the lockup with (0). maybe add the above code as a simple example of SpeedControl

Well, 1-2ms is not really fast for a 120MHz CPU. Just don't use it in a tight loop. Anyway, the default update rate of the motor speed is 5ms. I.e. the motor speed won't change more often than once per 5ms...

> I know you wrote the code to synchronize pumps etc. but I think a few of us in the forums are trying to use your code for 4 independent controllable pumps, for instance I am trying to use it to [...]

Sounds like a very interesting project. Do you happen to have a link to more information about it?  4 independent motors should work without issue. Above 4 it gets difficult since the Teensy only has 4 of the required hardware timers.

closed 2021-11-25 by luni64

EeeLo, 2023-02-23:
I using this fix version and also not start motor after overrideSpeed(0), but I'm using STM32F4 board

commit 12754333 (2021-11-25): Fix bug preventing restart after overrideSpeed(0)

Fixes #119