luni64/TeensyStep #121: The teensyStep controller.moveAsync() get stuck after two to three movements. I have also tried to use while(controller.isRunning); But it is still getting stuck. I tried adding controller.stopAsync() after while(controller.isRunning); but ended up with same result. Please help.

prajesh-asink, 2021-12-02:


luni64, 2021-12-02:
Please post a minimal example showing the effect.

prajesh-asink, 2021-12-02:
while (controller.isRunning())
      ;
      controller.stopAsync();
      
      m1.setTargetAbs(data[0]);
      m2.setTargetAbs(data[1]);
      m3.setTargetAbs(data[2]);
      m4.setTargetAbs(data[3]);
      m5.setTargetAbs(data[4]);
      m6.setTargetAbs(data[5]);
      controller.moveAsync(m1, m2, m3, m4, m5, m6);

My code keeps sending values to populate the data[] array over serial communication for a pi to teensy 3.5. But after 2 to 3 such runs it just get stuck.

luni64, 2021-12-02:
Thanks for the snippet. Unfortunately it doesn't show much.  If you want me to help you I really need a complete sketch which I can compile and test. The sketch needs to demonstrate the issue without need for additional hardware / software.

prajesh-asink, 2021-12-02:
Thank you so much for the prompt replies. It seems I just had to add a little delay in the while loop,

```
while (controller.isRunning())
{
     delay(1);
}
```

it is working now.
Thank you very much for your amazing work!

luni64, 2021-12-02:
Great. 
BTW: instead of you while loop you could simply do a controller.stop().  This has the same effect as you programmed manually

prajesh-asink, 2021-12-02:
Thanks anyway 🥂

closed 2021-12-02 by luni64