luni64/TeensyStep #122: Trying to Understand the "TIM[x] Not Declared" for STM32F411CE Black Pill

Kuantronic, 2021-12-30:
Hi @ramboerik . Before I get into the small issue & questions, I got to say that I really appreciate your work in integrating the STM32 into the TeensyStep library!

I made an example test sketch for the WeAct Black Pill STM32F411CE to use this library. When I compiled, I got this error below

```
TimerField.cpp: 4:80: error: 'TIM6' was not declared in this scope; did you mean 'TIM1'?
   timer_mapping[MAX_TIMERS] = { TIM1, TIM2, TIM4, TIM5, TIM6, TIM7, TIM8, TIM9, TIM10, TIM11, TIM12, TIM14 }
   |                                                                                ^~~~
   |                                                                                TIM1
 
TimerField.cpp: 4:86: error: 'TIM7' was not declared in this scope; did you mean 'TIM1'?
   timer_mapping[MAX_TIMERS] = { TIM1, TIM2, TIM4, TIM5, TIM6, TIM7, TIM8, TIM9, TIM10, TIM11, TIM12, TIM14 }
   |                                                                                      ^~~~
   |                                                                                      TIM1
 
TimerField.cpp: 4:92: error: 'TIM8' was not declared in this scope; did you mean 'TIM1'?
   timer_mapping[MAX_TIMERS] = { TIM1, TIM2, TIM4, TIM5, TIM6, TIM7, TIM8, TIM9, TIM10, TIM11, TIM12, TIM14 }
   |                                                                                            ^~~~
   |                                                                                            TIM1
 
TimerField.cpp: 4:118: error: 'TIM12' was not declared in this scope; did you mean 'TIM1'?
   timer_mapping[MAX_TIMERS] = { TIM1, TIM2, TIM4, TIM5, TIM6, TIM7, TIM8, TIM9, TIM10, TIM11, TIM12, TIM14 }
   |                                                                                                                      ^~~~~
   |                                                                                                                      TIM1
 
TimerField.cpp: 4:125: error: 'TIM14' was not declared in this scope; did you mean 'TIM1'?
   timer_mapping[MAX_TIMERS] = { TIM1, TIM2, TIM4, TIM5, TIM6, TIM7, TIM8, TIM9, TIM10, TIM11, TIM12, TIM14 }
   |                                                                                                                             ^~~~~
   |                                                                                                                             TIM1
```

I did go into the TimerField.cpp and remove those TIM[x] in the array that had the error message and that solved the problem. It is because these timer numbers don't exist for the specific 411CE MCU model? If so, is there a way to have different timer_mapping arrays used depending on the model of MCU selected?

I'm very new to this low level of Arduino programming so I'm trying to learn as I go.

ramboerik, 2022-01-03:
Hi @Kuantronic, nice to hear it works for you! Yes you are correct, the number of hardware timers differ between the stm32 models. I originally ajdusted the timers for `STM32F429` but a more robust auto-detect should have been implemented from the start 😃    
 I guess a straight forward soultion for the mapping is to check the actual processor define (such as `STM32F429`) and make a custom variable declaration for each model.

Kuantronic, 2022-01-12:
Thanks for the info! I thought I replied when I first saw this but apparently I didn't so sorry for the late response. :/

I might play around with the file on my end and see if I can add different timer arrays for the different models. This might allow it for auto detect in the future.

closed 2022-01-12 by Kuantronic