luni64/TeensyStep #123: Invoke callback from stopAsync() as well?

bimac, 2022-01-24:
Hi Lutz,

thanks for your work on this excellent library! Would it be possible to invoke the callback function (`setCallback()`, cf. #18) from other async methods as well? I'm trying to execute code right _after_ the motor has slowed to a stop with `stopAsync()`, ideally without polling `isRunning()` ...

Thanks!!
Florian

luni64, 2022-01-24:
Hi Florian, 

I don’t quite understand. The callback function is invoked by TeensyStep not by some user function. Once set it will be called whenever the controller finished the movement. (However, this is a seldom used functionality. IIRC I never got any questions about it… so, chances are that something didn't make it through all the updates) 

Just give it a try.

bimac, 2022-01-24:
Uhm -- - yes. Apologies. It's been a long day. I got confused by my own code -- let's ignore what I wrote above :-)

I am using, both, `StepControl` and `RotateControl` for invoking motions with or without a predefined target, respectively (see [here](https://github.com/poulet-lab/Bpod_Stepper/blob/c8bdd5149d12980204a4bba20f82e92ca5ffd8d7/firmware/Bpod_stepper/StepperWrapper_TeensyStep.cpp#L33-L36)). `StepControl` implements `setCallback()` and it works fine. But there is no such function for the async Methods in `RotateControl`, correct? So if I end a rotation using `stopAsync()` I currently can't trigger a callback at the end of the deceleration-phase. I'll correct the title - sorry for the confusion.

luni64, 2022-01-26:
Yes, the callback mechanism was only implemented for the step controllers, not for the rotate controllers. I meanwhile added the feature and uploaded it to the RotateCallback branch. I currently don't have a lot of time for testing. May I ask you to carefully check if it works? I'll push it to the main branch afterwards. 

Here what I tried: 
```c++
#include "Arduino.h"
#include "TeensyStep.h"

RotateControl rc;
Stepper stepper(0, 1);

void setup()
{
  while(!Serial){}
  Serial.println("start");

  rc.setCallback([] { Serial.println("stop"); });

  rc.rotateAsync(stepper);
  delay(1000);
  rc.stopAsync();
}

void loop(){
}
```

bimac, 2022-01-27:
Thank you!! Yes, happy to give it a spin. I can see the branch but it seems to be identical with master - did you perhaps forget to push the changes? Cheers.

luni64, 2022-01-27:
That's true indeed. Should be pushed now...

bimac, 2022-01-28:
I just played around with it and it looks good. The callback reliably trigger at the end of the deceleration for both `StepControl` and `RotateControl`. Is there anything else I should test?

luni64, 2022-01-28:
Thanks for bringing that up, merged into master

PR #126 (2022-01-28): Added callback functionality to RotateControl
Fixes #123
closed 2022-01-28 by luni64