luni64/TeensyStep #128: TeensyStep stops working if you use it somewhere TIM13 or use TIM2, TIM5 as stepTimer

duritskiy, 2022-02-02:
I've extended the TeensyStep library a bit. Added timer enforcement to TeensyStep::StepControlBase. Now it might look like
TeensyStep::StepControlBase<LinStepAccelerator, TimerField> StepController1(6,5000,TIM6,TIM5,TIM7 );

And here's what I found out: For reasons I don't understand, TeensyStep stops working if you use it somewhere
TIM13 or use TIM2, TIM5 as stepTimer. In all other cases it seems to work.

Do you know the reason for these bugs?

P.S.
Please add my addition to your library. I see them as very useful.
Thank you.
[stm32_TeensyStep_Encoders.zip](https://disk.yandex.ru/d/AiwqXdqiG-mOiQ)

ramboerik, 2022-02-02:
Hi @duritskiy! 

Just out of curiosity, what board are you using? 😃 I had problems with TIM13 and TIM3 with the stm32f429/stm32f4f05 and as far as I remember at least one was used by the underlaying stm32 core library.  

Regarding the code, do your changes also compile for the teensy board?

duritskiy, 2022-02-03:
[env:STM32F407VET6]
platform = ststm32@~13.0.0
board = black_f407ve

https://github.com/mcauser/MCUDEV_DEVEBOX_F407VET6

The changes affected only three constructors.
If the same is changed for Teensy, then it will also work.

_**\lib\TeensyStep\src\StepControlBase.h**_
StepControlBase::StepControlBase(unsigned pulseWidth = 5, unsigned accUpdatePeriod = 5000,TIM_TypeDef *_stepTimer=0,TIM_TypeDef *_accTimer=0,TIM_TypeDef *_pulseTimer=0)

_**\lib\TeensyStep\src\MotorControlBase.h**_
MotorControlBase::MotorControlBase(unsigned pulseWidth, unsigned accUpdatePeriod,TIM_TypeDef *_stepTimer=0,TIM_TypeDef *_accTimer=0,TIM_TypeDef *_pulseTimer=0) 

_**\lib\TeensyStep\src\timer\stm32\TimerField.cpp**_
TimerField::TimerField(TeensyStep::TF_Handler *_handler,TIM_TypeDef *_stepTimer,TIM_TypeDef *_accTimer,TIM_TypeDef *_pulseTimer)
    : 
		stepTimer( _stepTimer?_stepTimer:get_timer()), 
		accTimer(_accTimer?_accTimer:get_timer()), 
		pulseTimer(_pulseTimer?_pulseTimer:get_timer()), 
		stepTimerRunning(false)
{
...
}

duritskiy, 2022-02-03:
I just looked at what Teensy is and realized that this is not STM32, but NXP.
Apparently it is also easy to remake the code will not work. Or maybe it is possible - I did not really delve into it.

ramboerik, 2022-02-03:
Yes, the teensy is quite different. I guess it's up to @luni64 to decide if the library should provide a way of specifying what interrupts to use. If the library would provide such a mechanism then I can add it to the stm32 parts 😃