luni64/TeensyStep #131: TeensyStep HelloStepper with Teensy 4.1 will not compile

Atomfusion1, 2022-03-13:
Hello, With what could be a shortage of Teensy 3.5/3.6 but PJRC saying that they have a lot of Teensy 4.1 available I am looking to move my code over to it, I see in some of your other projects you have already implemented Teensy 4.0 and 4.1 code, but the Current TeensyStep HelloStepper will not compile under 4.1. I will assume the hardware timers are not in it yet, do you happen to have a development branch for them yet ? 

`In file included from F:\PCloud\arduino-1.8.19\libraries\TeensyStep\src/timer/generic/TimerField.h:6:0,
                 from F:\PCloud\arduino-1.8.19\libraries\TeensyStep\src/TeensyStep.h:11,
                 from F:\PCloud\arduino-1.8.19\libraries\TeensyStep\examples\HelloStepper\HelloStepper.ino:19:
F:\PCloud\arduino-1.8.19\libraries\TeensyStep\src/timer/generic/TickTimer.h: In static member function 'static uint32_t TimerBase::get_cycles()':
F:\PCloud\arduino-1.8.19\libraries\TeensyStep\src/timer/generic/TickTimer.h:28:34: error: 'dwt_getCycles' was not declared in this scope
             return dwt_getCycles();
`

Thanks

christophepersoz, 2022-03-29:
Hello, same here. I would be tremendous to get an update for Teensy 4.x boards as they are much more powerful compare to 3.6. Is there any plan to do so or just out of the scope?
Thanks a lot for your amazing lib.

C

luni64, 2022-03-29:
There is a very first experimental version here: 
https://github.com/luni64/TeensyStep4

ipsod, 2022-03-31:
Glad to see this is happening.  Teensy 3 boards are out of stock.

christophepersoz, 2022-05-04:
> There is a very first experimental version here:
> https://github.com/luni64/TeensyStep4

Nice, thank a lot. Is this version working? I do not have Teensy 4.x board to test it. Thanks

luni64, 2022-05-05:
It should be working in principle (used it in a small project) but there certainly will be bugs.

KrisKasprzak, 2022-09-01:
I downloaded and installed this lib but get

C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_957164\sketch\main.cpp:2:25: fatal error: teensystep4.h: No such file or directory
compilation terminated.
Error compiling for board Teensy 4.0.

Any help appreciated.

luni64, 2022-09-02:
I'd like to reproduce that. Which sketch did you try to compile? Which build system are you using (Arduino/PIO/VisualTeensy...)?

KrisKasprzak, 2022-09-04:
silly me...

1. sketch: 01_basic.ino
2. install path:  C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\TeensyStep4-main\examples\01_basic
3. Win10
4. IDE Arduino 1.8.13
5. Teensy loader 1.57

I have no extra compilers, add-on, I run basic Arduino IDE with Teensyduino.

I did move all .h and .cpp files to the TeensyStep4-main\ folder and changed all the include path and I'm able to compile. Just odd my Arduino could not find all the files on all the sub-folders.

TIA