luni64/TeensyStep #132: fatal error: algorithm: No such file or directory

lebadu, 2022-04-26:
We are having problems adding the TeensySteper library to the Arduino IDE. The error is described in the image below. When I try to add the algorithm library as shown below it keeps giving the error adding other libraries which are not available for use. can you help us to fix it
Thank you very much. 

[image]

luni64, 2022-04-26:
The lib only works for Teensy 3x. You compiled for Mega

lebadu, 2022-04-26:
Did you mean that Lib only works when using the Teensy USB Development Board electronic circuit, it can't work directly on the Arduino Mega2560 board right?

luni64, 2022-04-26:
Yes, only for Teensy 3.x as written in the library description: 

[image]

closed 2022-04-28 by luni64