luni64/TeensyStep #133: Multiple Stepper Synchronization with new position control at specified speed

harshdobariya, 2022-05-28:
First of all I am sorry for rising this issue here it might not belong here and I would close it soon. But I am stuck here and would really like have your opinion. 

So I have build a 6-axis manipulator having all the degree of freedom. Currently I have used TeensyStep library for controlling synchronization of 6 stepper axis. Until now everything works great.

But now I have to move the robot (all 6-axis), with the help of numeric control code.
(Basically what happens in 3D printer control with Gcode)
For example :  
```
N001   A1 150,  A2 32.6, A3 -50, A4 -75.8, A5 50, A6 10.2
N002   A1 120,  A2 42.6, A3 -55.0, A4 88, A5 150, A6 10.8
N003   A1 50,  A2 3.6, A3 -5.50, A4 -5.8, A5 150, A6 1.2
N004   A1 158,  A2 38, A3 80, A4 7.8, A5 44.4, A6 100.2
```
where A1 to A6 represents angle values in degree. 

With AccelStepper or TeensyStep library I can move to give code line(N00x) in a synchronous move. So when I feed this code lines it will go from current position accelerating-to maxSpeed- then deaccelerating to target position.

What I want to do is to move from one code line to next code line and to next in One-Continuous move. That is no accelerating from start position to setSpeed/maxSpeed to first point - then to next point - then to next point and deaccelerating finally at the end of code line.
So if there is new code line in the buffer it will not deaccelerate, go to current target with maxSpeed/setSpeed and then to new target with updated speed and so on.

Basically what Computer Numeric Control(CNC) machines do.

One of the suggestion I got is that this problem requires "look ahead motion planning".
So is there any way to modify TeensyStep library so that I can achieve this task?

I tried searching regarding this in many different platforms but can't find anything closely related to this.

atlesg, 2022-06-06:
the **blended motion / continuous movement / look ahead motion planning** is one tough topic for every motion controller developers. I don't think TeensyStepper has this function, I would love to donate, theoretically support for author's team to add this one.

BlueGene00, 2022-07-27:
This topic is exactly what I want to achieve. (And now have achieved)

Please have a look here:
https://github.com/luni64/TeensyStep/issues/139

This is basically all you need. You just need to calculate the positions and speed before and save them in arrays. Then you can use many tiny linear segments to interpolate any desired movement.

I will perform some more detailed tests and then publish a complete documentation with some examples.

For my movements I use bezier-splines, so it will be possible to just hand over a few positions and everything in between will be calculated with a smooth curve.

I tried Accelstepper before, but it was not capable of running multiple steppers with high step frequency. The first tests with TeensyStep and the adjusted code above look very promising. The only open topic after that would be the same implementation to the Teensy 4, because Teensy 3.6 will not be available for a couple more months.