luni64/TeensyStep #137: Supported STM32 boards

JeppeKlitgaard, 2022-07-12:
Will be updated in a minute

luni64, 2022-07-12:
The STM32 version lives in the branch STM32. Did you have a look there? Your PR seems to address the main branch. 

BTW: the STM32 branch is maintained by @ramboerik. Actually I currently don't use the STM boards and have not much knowledge of them.

JeppeKlitgaard, 2022-07-13:
@luni64 I completely missed that there was a separate branch for STM32. Wouldn't it be possible to merge that functionality into the `master` branch, gating it with a `#if defined(...)`?

Perhaps you could reference the STM32 branch in the `README.md`, as it is quite easy to miss as is. Thanks for making TeensyStep, I should add. It is really nice to use!

Feel free to close this issue