luni64/TeensyStep #140: Does this work with the TeensyLC (MKL26Z64)

payor-ma, 2022-08-05:
Hello, just wondering if this works with the teensyLC boards?  I had a brief scroll through the timer stuff and it looks like at this stage there's a generic implementation through the Arduino standard libraries, and an ARM-specific implementation that's caught by the teensy3.2-4 processor flags.  So I would guess that the teensyLC would compile to use the generic timer implementation thru Arduino.h.  Is there a performance hit with that vs the ARM-specific implementation?

luni64, 2022-08-05:
The used timers are specific to the processor. Unfortunately the LC has a limited set of timers so that won't work out of the box. It might be possible to do something for the LC but frankly, this is not on my priority list

payor-ma, 2022-08-06:
That's no problem.  I'm looking to run one motor with a teensyLC thru either a step/dir interface or a uart interface (with a bit of hardware hacking), and I was considering my options.  If you're happy to answer some more questions:
From what I've read the TeensyLC's PWM timers are TPM not FTM, and it has 2 instead of 4 PIT timers.  It looks like a lot of the FTM timer code has catches for using TPM or FTM timers, are there any capabilitiy issues using TPM?  Briefly looking through the repo it looks like the 4 PIT timers are coded in for the teensy implementations of the timers.  What are these 4 used for and would it be possible to use only 2?

closed 2022-08-16 by payor-ma