luni64/TeensyStep #144: Arduino due

vp8181, 2022-10-01:
Does this library work for Arduino due? I just want to control 6 stepper at 1600 steps(1000rpm).

Also any updates of the teensystep4 Library....

vp8181, 2022-10-07:
Hello,
Has anyone tried this library with Arduino due.
I would like to buy one if it works. Please help 🙏🙏🙏
Thanks in advance.
@luni64  please let me know sir

luni64, 2022-10-07:
No this library only works with T3.x boards

vp8181, 2022-10-07:
STM32F407VET6 ARM CORTEX -M4 

https://robu.in/product/stm32f407vet6-arm-cortex-m4-core-with-dsp-and-fpu/?pid=835836

What about this? Should i buy this? 
As i have seen many error are there for this in the issues section ... What about now? Is it stable?

vp8181, 2022-10-08:
Can anyone please confirm it .Its urgent as only few are left and need to complete the project.
Please @luni64 @ramboerik

ramboerik, 2022-10-09:
Hi,

That board should run it. It is the same family as the one I developed it for: https://www.adafruit.com/product/4382.  Just rememberer to use the stm32 branch of the library.

vp8181, 2022-10-09:
@ramboerik  thankyou so much 🙏
@luni64 thanks

closed 2022-10-10 by vp8181