luni64/TeensyStep #151: Issue compiling with Teensyduino 1.58 on PIO

bimac, 2023-04-06:
Hi luni64,

I can't seem to compile TeensyStep with Teensyduino 1.58 on PlatformIO:

`platformio.ini`:
```ini
[env:teensy35]
platform = teensy
framework = arduino
board = teensy35
lib_deps = https://github.com/luni64/TeensyStep.git
```

log output:
```
In file included from /home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/WProgram.h:46,
                 from /home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/Arduino.h:6,
                 from .pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:1:
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp: In member function 'TeensyStep::Stepper& TeensyStep::Stepper::setPullInSpeed(int32_t)':
/home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/wiring.h:141:16: error: expected unqualified-id before '(' token
  141 | #define abs(x) ({ \
      |                ^
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:90:35: note: in expansion of macro 'abs'
   90 |         vPullIn = vPullOut = std::abs(speed);
      |                                   ^~~
/home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/wiring.h:142:3: error: expected primary-expression before 'typeof'
  142 |   typeof(x) _x = (x); \
      |   ^~~~~~
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:90:35: note: in expansion of macro 'abs'
   90 |         vPullIn = vPullOut = std::abs(speed);
      |                                   ^~~
/home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/wiring.h:142:3: error: expected '}' before 'typeof'
  142 |   typeof(x) _x = (x); \
      |   ^~~~~~
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:90:35: note: in expansion of macro 'abs'
   90 |         vPullIn = vPullOut = std::abs(speed);
      |                                   ^~~
/home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/wiring.h:141:17: note: to match this '{'
  141 | #define abs(x) ({ \
      |                 ^
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:90:35: note: in expansion of macro 'abs'
   90 |         vPullIn = vPullOut = std::abs(speed);
      |                                   ^~~
/home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/wiring.h:142:3: error: expected ')' before 'typeof'
  142 |   typeof(x) _x = (x); \
      |   ^~~~~~
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:90:35: note: in expansion of macro 'abs'
   90 |         vPullIn = vPullOut = std::abs(speed);
      |                                   ^~~
/home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/wiring.h:141:16: note: to match this '('
  141 | #define abs(x) ({ \
      |                ^
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:90:35: note: in expansion of macro 'abs'
   90 |         vPullIn = vPullOut = std::abs(speed);
      |                                   ^~~
/home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/wiring.h:143:4: error: '_x' was not declared in this scope
  143 |   (_x > 0) ? _x : -_x; \
      |    ^~
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:90:35: note: in expansion of macro 'abs'
   90 |         vPullIn = vPullOut = std::abs(speed);
      |                                   ^~~
/home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/wiring.h:144:1: warning: no return statement in function returning non-void [-Wreturn-type]
  144 | })
      | ^
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:90:35: note: in expansion of macro 'abs'
   90 |         vPullIn = vPullOut = std::abs(speed);
      |                                   ^~~
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp: At global scope:
/home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/wiring.h:144:2: error: expected unqualified-id before ')' token
  144 | })
[... 80 lines cut ...]
/home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/wiring.h:143:4: error: '_x' was not declared in this scope
  143 |   (_x > 0) ? _x : -_x; \
      |    ^~
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:111:18: note: in expansion of macro 'abs'
  111 |         A = std::abs(delta);
      |                  ^~~
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp: At global scope:
/home/bimac/.platformio/packages/framework-arduinoteensy/cores/teensy3/wiring.h:144:2: error: expected unqualified-id before ')' token
Compiling .pio/build/teensy35/FrameworkArduino/HardwareSerial2.cpp.o
  144 | })
      |  ^
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:111:18: note: in expansion of macro 'abs'
  111 |         A = std::abs(delta);
      |                  ^~~
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:112:5: error: expected declaration before '}' token
  112 |     }
      |     ^
.pio/libdeps/teensy35/TeensyStep/src/Stepper.cpp:113:1: error: expected declaration before '}' token
  113 | }
      | ^
Compiling .pio/build/teensy35/FrameworkArduino/HardwareSerial3.cpp.o
*** [.pio/build/teensy35/liba5a/TeensyStep/Stepper.cpp.o] Error 1
```

Cheers

luni64, 2023-04-07:
Should be fixed in v2.3.3

bimac, 2023-04-07:
Indeed it is :) thanks for the fix!

closed 2023-04-07 by bimac