luni64/TeensyStep #152: RotateController with overridespeed sometime delay

EeeLo, 2023-04-12:
I'm using RotateController to controll stepping motor and overridespeed to change speed, but there have delay time to run motor sometimes, I don't know why, some good idea or any questions befor?

EeeLo, 2023-04-13:
As first thank you for your wonderful lib and good job on it!
I using your lib to control a turntable with RotateController overrideSpeed function and found that my code set it but the motor or getPosition() shows some delay after that sometimes, it's random not everytime. I'm run my code on Arduino IDE and STM32 F446 board by your PID(PI) control code in loop().

```
void motor_run(double az_move, double el_move, double xy_move, int angle_x)
{
    //static unsigned lastTick = 0; 
    if(power_tmp != 0) return;

    
    delta_az = (((az_move + revise_az) * az_factor * AZ_DIR) - az.getPosition()) * (P_AZ / PID_AZ);                // This implements a simple P regulator (can be extended to a PID if necessary)
    delta_el = (((el_move + revise_el) * el_factor * EL_DIR) - el.getPosition()) * (P_EL / PID_EL);                // This implements a simple P regulator (can be extended to a PID if necessary)

    factor_az = std::max(-1.0f, std::min(1.0f, delta_az));       
    factor_el = std::max(-1.0f, std::min(1.0f, delta_el));         

    controller_az.overrideSpeed(factor_az);       
    controller_el.overrideSpeed(factor_el);      

   Serial.println(String(factor_az) +" " + String(get_az_angle()));
    delay(3);
}
```

The overrideSpeed in the loop less than 1ms, but I already test as more than 10ms.
this is my print data:

factor     angle
0.01 109.59
0.01 109.59
0.01 109.59..
0.01 109.59..
0.01 109.59..0.01 109.59..
0.01 109.59..
0.01 109.59..0.01 109.59..0.00 109.59..
-0.01 109.58..-0.01 109.58..-0.01 109.58..
-0.01 109.58..
-0.01 109.58..-0.02 109.58..-0.02 109.58..
-0.02 109.58..
-0.02 109.58..-0.02 109.58..
-0.02 109.58..
-0.02 109.58..-0.03 109.58..-0.03 109.58..
-0.03 109.58..
-0.03 109.58..-0.04 109.58..
-0.04 109.58..
-0.04 109.58..-0.06 109.58..
-0.06 109.58..
-0.06 109.58..-0.07 109.58..-0.07 109.58..
-0.07 109.58..
-0.07 109.58..-0.09 109.58..
-0.09 109.58..
-0.09 109.58..
-0.12 109.58..-0.12 109.58..
-0.12 109.58..
-0.14 109.58..-0.14 109.58..-0.14 109.58..
-0.14 109.58..-0.17 109.58..
-0.17 109.58..
-0.17 109.58..-0.20 109.58..
-0.20 109.58..
-0.20 109.58..-0.20 109.58..-0.23 109.58..
-0.23 109.58..
-0.23 109.58..-0.27 109.58..
 -0.27 109.58..
-0.27 109.58..-0.30 109.58..-0.30 109.58..
 -0.30 109.58..
 -0.30 109.58..-0.34 109.58..
 -0.34 109.58..
-0.34 109.58..-0.38 109.58..
-0.38 109.58..
-0.38 109.58..-0.38 109.58..
-0.43 109.58..
-0.43 109.58..
-0.43 109.58..-0.47 109.58..
-0.47 109.58..-0.47 109.58..
-0.52 109.58..-0.52 109.58..
-0.52 109.58..
-0.52 109.58..-0.57 109.58..
-0.57 109.58..-0.57 109.58..
-0.63 109.58..
-0.63 109.58..-0.63 109.58..-0.68 109.58..
-0.68 109.58..-0.68 109.58..
-0.68 109.58..
-0.75 109.58..-0.75 109.58..-0.75 109.58..
-0.81 109.58..-0.81 109.58..
-0.81 109.58..-0.81 109.58..
-0.89 109.58..-0.89 109.58..
-0.89 109.58..
-0.96 109.58..-0.96 109.58..
-0.96 109.58..
-1.00 109.58..-1.00 109.58..-1.00 109.58..
-1.00 109.58..-1.00 109.58..
-1.00 109.58..
-1.00 109.58..-1.00 109.58..
-1.00 109.58..
-1.00 109.58..-1.00 109.58..
-1.00 109.58..
-1.00 109.58..-1.00 109.58..
-1.00 109.58..-1.00 109.58..
-1.00 109.58..-1.00 109.58..
-1.00 109.59..
-1.00 109.65..-1.00 109.70..-1.00 109.76..
-1.00 109.81..-1.00 109.87..
-1.00 109.92..
-1.00 109.97..-1.00 110.03..
-1.00 110.08..
-1.00 110.14..-1.00 110.19..
-1.00 110.24..
-1.00 110.30..
-1.00 110.35..-1.00 110.41..
-1.00 110.46..-1.00 110.51..-1.00 110.57..
-1.00 110.62..

the time market show that delay time around 0.5s

EeeLo, 2023-04-13:
sorry for my code.

I'm using stm32 board to run STM32 branches(Master doesn't work),but the problem is the code not restart after overrideSpeed(0),like this question: https://github.com/luni64/TeensyStep/issues/119#issuecomment-1441521493
so I try to fix it, I found this function in src/timer/stm32/TimerField.h
void TimerField::setStepFrequency(unsigned f)
```
{
  //f == 0 ? end() : stepTimer.setOverflow(f, HERTZ_FORMAT);
  stepTimer.setOverflow(f, HERTZ_FORMAT);
}
```
I dumped to end() and the motor can running with overrideSpeed()
but the new problem is when my code read new angle data from sensor,my stepper motor fix angle by overrideSpeed, but there have delay sometimes, I print the overrideSpeed factor and can see that the code already call overrideSpeed but getPosition() not changed timely sometimes.
so:
1, why the master doesn't work on my stm32 board?
2, why motor delay sometimes?

I'm a rookie,thanks.

EeeLo, 2023-04-13:
```
#include <time.h>
#include <stdio.h>
#include <Arduino.h>
#include <TeensyStep.h>

#define az_factor  12.222
#define AZ_PUL_PIN    PB3
#define AZ_DIR_PIN    PB4

static double gotoangle_az;
float delta_az = 0;

Stepper az(AZ_PUL_PIN, AZ_DIR_PIN); 
RotateControl controller_az(5, 500); 

void setup() {
    az.setMaxSpeed(AZ_MAX_SPEED)             // stp/s
      .setAcceleration(AZ_ACCELERATION);     // stp/s^2
    controller_az.rotateAsync(az);
    controller_az.overrideSpeed(0);
}

void motor_run(double az_move)
{
    delta_az = ((az_move * az_factor ) - az.getPosition()) * (P_AZ / PID_AZ);  

    factor_az = std::max(-1.0f, std::min(1.0f, delta_az));         

    controller_az.overrideSpeed(factor_az);       

   Serial.println(String(factor_az) +" " + String(get_az_angle()));
    delay(3);
}

void loop() {

  //gotoangle_az data from sensor 
  motor_run(gotoangle_az);
}
```

EeeLo, 2023-04-13:
I see, the first question is no setpTimerStart();
https://github.com/luni64/TeensyStep/commit/f4241d0f20d43592d625d9202a49a4909deac5f6

luni64, 2023-04-13:
Thanks for providing the script. I assume that this script will generate the issue of delays right? 

TeensyStep is developed for Teensy 3.x boards. @ramboeric added support for stm32 boards in the STM32 branch of this library. 

I'll check if I can reproduce the problem with a Teensy 3.6 using your skript. If this is STM related I can't help you.

luni64, 2023-04-13:
Sorry, your script does not compile, (factor_az not defined). Are you sure that this is a a script which will generate the problem??

EeeLo, 2023-04-15:
Thanks for your mind, I find the problem maybe is from stm32duino library's HardwareTimer or my stm32 board hardware.

EeeLo, 2023-04-15:
This is test code to run motor with RotateController,problem is the stm32 board freeze sometimes when change direction.

#include <time.h>
#include <stdio.h>
#include <TeensyStep.h>
#include <elapsedMillis.h> 

#define az_factor  55.555
#define AZ_MAX_SPEED 2000
#define AZ_ACCELERATION 4000

unsigned long timenow = 0;

#define AZ_PUL_PIN    PB3
#define AZ_DIR_PIN    PB4

Stepper az(AZ_PUL_PIN, AZ_DIR_PIN);
RotateControl controller_az(5, 500);

void setup() {
      Serial.begin(230400);               
      az.setMaxSpeed(AZ_MAX_SPEED)       
        .setAcceleration(AZ_ACCELERATION);   
      controller_az.rotateAsync(az);
      controller_az.overrideSpeed(0);

      int dir = 1;
      while(1){
        if(dir == 1)
        {
            controller_az.overrideSpeed(dir);
            Serial.print("overrideSpeed: 1");
            delay(1000);
            dir = -1;
        }else{
            controller_az.overrideSpeed(dir);
            Serial.println("overrideSpeed: -1");
            delay(1000);
            dir = 1;
        }
      };
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

EeeLo, 2023-04-16:
My problem is very like this https://github.com/luni64/TeensyStep/issues/47#issue-443701798 
At first the motor doesn't runing after start,so I hidden the end() code:
```
void TimerField::setStepFrequency(unsigned f)
{
  //f == 0 ? end() : stepTimer.setOverflow(f, HERTZ_FORMAT);
  //stepTimer.setOverflow(f, HERTZ_FORMAT);

  if(f == 0)
  {
      //end();
      //Serial1.println("speed is 0");
  }else{
      stepTimer.setOverflow(f, HERTZ_FORMAT);
  }
}
```
but motor always freeze sometimes when I changed direction.

EeeLo, 2023-07-10:
> 我的问题很像[#47（评论）](https://github.com/luni64/TeensyStep/issues/47#issue-443701798) 首先电机在启动后不运行，所以我隐藏了 end() 代码：
> 
> ```
> void TimerField::setStepFrequency(unsigned f)
> {
>   //f == 0 ? end() : stepTimer.setOverflow(f, HERTZ_FORMAT);
>   //stepTimer.setOverflow(f, HERTZ_FORMAT);
> 
>   if(f == 0)
>   {
>       //end();
>       //Serial1.println("speed is 0");
>   }else{
>       stepTimer.setOverflow(f, HERTZ_FORMAT);
>   }
> }
> ```
> 
> 但当我改变方向时，电机有时总是冻结。

I find out what is my problem is. the STM32F4 branch can't restart motor after overrideSpeed(0) and I change the lib code to fix it, but the timer is not end when speed set 0 or 1, the timer period ARR is too slowly to update new speed value, that is why motor delay some time after restart.
[https://github.com/luni64/TeensyStep/issues/157](url)