luni64/TeensyStep #155: The analog Joystick code doesn't work on the STM32F4

chihoonl, 2023-05-06:
First, I'd like to thank Luni64 for letting me use this awesome library.
Sorry about that I don't know whether this is an issue with the library or a minor mistake in my code. But the code is fully tested on Teensy 3.xx. But it doesn't work on STM32F4(Nucleo F446re, Nucleo F429zi. I tried on F411, F401 also.)
Unfortunately, I'm a newbie to STM32, so I don't know what I'm missing. (Oddly enough, when I tested simply Teensystep's classes on STM32F4  without using the joystick, most of it worked.)
Teensy 3.xx is discontinued, and I have no choice except to port my longtime projects with Teensystep to STM32. Would anyone please advise me?

```
#include "Arduino.h"
#include "TeensyStep.h"

const int JOY_X_PIN = A0;
const int MOTOR_STEP_PIN = 3;
const int MOTOR_DIR_PIN = 2;
const int MAX_SPEED = 18000;
const int ACCELERATION = 9000;

Stepper motor(MOTOR_STEP_PIN, MOTOR_DIR_PIN);
RotateControl rotateControl;

void setup() {
  motor.setMaxSpeed(MAX_SPEED);
  motor.setAcceleration(ACCELERATION);
  rotateControl.rotateAsync(motor);
  rotateControl.overrideSpeed(0);
}

float transformPot(float analogVal) {
  if (analogVal > 550) {
    return map(analogVal, 550, 1023, 0.0f, 1.0f);
  } else if (analogVal < 420) {
    return map(analogVal, 0, 420, -1.0f, 0.0f);
  } else {
    return 0.0f;
  }
}

void loop() {
  static float oldOvrX = 0.0f;
  float ovrX = transformPot(analogRead(JOY_X_PIN));
  if (oldOvrX != ovrX) {
    oldOvrX = ovrX;
    rotateControl.overrideSpeed(ovrX);
  }
} 
```

EeeLo, 2023-07-10:
The lib is working on the STM32F4 well, I'm running with Nucleo F446RE board,the only problem is the RotateConteoller can't restart after overridespeed(0), I change the code to do that but not prefect.
/src/timer/stm32/TimerField.h
`
void TimerField::setStepFrequency(unsigned f)
{
  if(f == 0){
    stepTimerStop();
    return;
  }
  stepTimer.setOverflow(f, HERTZ_FORMAT);
}
`
to 

`void TimerField::setStepFrequency(unsigned f)
{
 /*
 if(f == 0){
    stepTimerStop();
    return;
  }
*/
  stepTimer.setOverflow(f, HERTZ_FORMAT);
}`

You can running on STM32F4 after this change,but you know, the stop cmd will not work,I'm researching it to find out another good way to fix it.

chihoonl, 2023-07-13:
I appreciated your great comment, I will test it again soon with this modified code! Thanks, EeeLo.

EeeLo, 2023-07-13:
> I appreciated your great comment, I will test it again soon with this modified code! Thanks, EeeLo.

You are wellcome, you also can use :if(f ==0) return;